wimaha/TeslaBleHttpProxy #2: Howto get

bill340, 2024-06-14:
I'm planning to implement your solution...
Just one question. From where / how can I get this data? I used teslalogger for it, but it also stopped working due to Tesla api limitations.

    soc:
      source: [Your Source ...]
    range:
      source: [Your Source ...]
    status:
      source: combined
      plugged:
        source: [Your Source ...]
      charging:
        source: [Your Source ...]

infinus, 2024-06-17:
I use teslamate and output the data to MQTT. I"m not sure about the 2nd part.

bill340, 2024-06-24:
As Teslamate and Teslalogger is still running into Fleet-API limits, could we not use the BLEProxy to get these parameters from the car?

wimaha, 2024-06-24:
I will update the documentation this week so that a detailed guide on how to handle these values will be available.
 
Unfortunately it is not possible to get these parameter via bluetooth from the car. This is currently not supported by Tesla. There is already a feature request on the vehicle command repository: https://github.com/teslamotors/vehicle-command/issues/52

Raudi1, 2024-07-04:
There's now a way to get limited data over BLE: https://github.com/teslamotors/vehicle-command/commit/05bc5dd8d0649b4ccb45a765b9127d06f1050a6f

wimaha, 2024-07-07:
But unfortunately no SOC over ble…
However since version 0.128.0 or newer of evcc it is very easy to integrate ble proxy. I’ve updated the readme.

closed 2024-07-07 by wimaha