wimaha/TeslaBleHttpProxy #3: Tesla M3 has no reaction on any click in evcc

pit65535, 2024-06-14:
I installed https://github.com/teslamotors/vehicle-command on Raspberry Pi 3 and from shell I can control my Tesla m3, so I istalled docker and copied the TeslaBleHttpProy Repo to /opt/TeslaBleHttpProxy and the private key to /opt/TeslaBleHttpProxy/TeslaBleHttpProxy/key/private.pem and run within this directory docker-compose up -d tesla-ble-http-proxy with the following docker-compose.yml:
version: '3.1'
services:
  tesla-ble-http-proxy:
    image: wimaha/tesla-ble-http-proxy
    container_name: tesla-ble-http-proxy
    volumes:
      - ~/TeslaBleHttpProxy/key:/key
      - /var/run/dbus:/var/run/dbus
    ports:
      - 8080:8080
    restart: always
    privileged: true
    network_mode: host
    cap_add:
      - NET_ADMIN
      - SYS_ADMIN

The Proxy starts and I can connect to port 8080 of the Raspberry´s IP.
Then I replaced my vehicle to the following definition:
vehicles:
  - name: Model_3
    type: custom
    title: Tesla Model 3
    capacity: 82
    chargeenable:
      source: http
      uri: "http://192.168.245.248:8080/api/1/vehicles/5YT3E7EB3MF944349/command/{{if .chargeenable}}charge_start{{else}}charge_stop{{end}}"
      method: POST
      body: ""
    maxcurrent: # set charger max current (A)
      source: http
      uri: http://192.168.245.248:8080/api/1/vehicles/5YT3E7EB3MF944349/command/set_charging_amps
      method: POST
      body: '{"charging_amps": "{{.maxcurrent}}"}'
    wakeup: # vehicle wake up command
      source: http
      uri: http://192.168.245.248:8080/api/1/vehicles/5YJ3ETEB3MF944349/command/wake_up
      method: POST
      body: ""
    soc:
      source: mqtt
      topic: teslamate/tesla/cars/1/battery_level
    range:
      source: mqtt
      topic: teslamate/tesla/cars/1/rated_battery_range_km
    status:
      source: combined
      plugged:
        source: mqtt
        topic: teslamate/tesla/cars/1/plugged_in
      charging:
        source: mqtt
        topic: teslamate/tesla/cars/1/charging_state

after restarting evcc (homeassistant-Plugin) it starts and log looks good. 
But if I try to start Charging by klicking at min+pv absolutely nothing happens, also teslamate still says the car is sleeping...
I then copied the private key also to /opt/TeslaBleHttpProxy/key/private.pem restarted the container, but still not working.
I tried again from shell: tesla-control -ble -vin 5YT3E7EB3MF944349 -key-file private.pem charging-start and it works.

I am not sure if it´s a bug or I did something wrong thats why I tryed to describe exactly what I did.

Any help would be very appreciated

wimaha, 2024-06-14:
Please try replacing `~/TeslaBleHttpProxy/key` with `/opt/TeslaBleHttpProxy/key/private.pem` in your docker-compose.yml.

And please provide the logs. You can get them by execute: `docker logs --since=1h tesla-ble-http-proxy`

bill340, 2024-06-14:
I have the same problem... Commands are not executed.
I installed everything according your docu.
If I try from cli
/home/uli/TeslaBleHttpProxy/key# tesla-control -ble -vin XP7YGCEK8RB277602 -key-file private.pem wake

I get this error:
Error: failed to find a BLE device: can't init hci: no devices available: (hci0: can't down device: device or resource busy)

It would be really great, if you could have a look to give me an idea, what I am doing wrong here... 

Here is the log:

2024/06/14 23:45:44 received command "{{if .chargeenable}}charge_start{{else}}charge_stop{{end}}" (VIN: XP7YGCEK8RB277602) with body: map[]
2024/06/14 23:45:44 The command "{{if .chargeenable}}charge_start{{else}}charge_stop{{end}}" is not supported.
2024/06/14 23:45:45 received command "wake_up" (VIN: XP7YGCEK8RB277602) with body: map[]
2024/06/14 23:45:45 handle command: wake_up (VIN: XP7YGCEK8RB277602)
2024/06/14 23:45:45 Failed to load private key: open key/private.pem: no such file or directory
2024/06/14 23:45:45 The command "wake_up" was canceled:
2024/06/14 23:45:45 [Error]Failed to load private key: open key/private.pem: no such file or directory
2024/06/14 23:47:04 received command "{{if .chargeenable}}charge_start{{else}}charge_stop{{end}}" (VIN: XP7YGCEK8RB277602) with body: map[]
2024/06/14 23:47:04 The command "{{if .chargeenable}}charge_start{{else}}charge_stop{{end}}" is not supported.
2024/06/14 23:47:05 received command "wake_up" (VIN: XP7YGCEK8RB277602) with body: map[]
2024/06/14 23:47:05 handle command: wake_up (VIN: XP7YGCEK8RB277602)
2024/06/14 23:47:05 Connecting to vehicle...
2024/06/14 23:47:08 sending command "wake_up"...
2024/06/14 23:47:08 The command "wake_up" was successfully executed.
2024/06/14 23:47:57 TeslaBleHttpProxy is loading ...
2024/06/14 23:47:58 TeslaBleHttpProxy is running!
2024/06/14 23:48:09 received command "{{if .chargeenable}}charge_start{{else}}charge_stop{{end}}" (VIN: XP7YGCEK8RB277602) with body: map[]
2024/06/14 23:48:10 The command "{{if .chargeenable}}charge_start{{else}}charge_stop{{end}}" is not supported.
2024/06/14 23:48:10 received command "wake_up" (VIN: XP7YGCEK8RB277602) with body: map[]
2024/06/14 23:48:11 handle command: wake_up (VIN: XP7YGCEK8RB277602)
2024/06/14 23:48:12 Connecting to vehicle...
2024/06/14 23:48:30 sending command "wake_up"...
2024/06/14 23:48:30 The command "wake_up" was successfully executed.
2024/06/14 23:48:41 received command "set_charging_amps" (VIN: XP7YGCEK8RB277602) with body: map[charging_amps:16]
2024/06/14 23:48:41 handle command: set_charging_amps (VIN: XP7YGCEK8RB277602)
2024/06/14 23:48:41 Connecting to vehicle...
2024/06/14 23:48:43 sending command "set_charging_amps"...
2024/06/14 23:48:44 The command "set_charging_amps" was successfully executed.
2024/06/14 23:49:44 received command "{{if .chargeenable}}charge_start{{else}}charge_stop{{end}}" (VIN: XP7YGCEK8RB277602) with body: map[]
2024/06/14 23:49:44 The command "{{if .chargeenable}}charge_start{{else}}charge_stop{{end}}" is not supported.
2024/06/14 23:49:44 received command "wake_up" (VIN: XP7YGCEK8RB277602) with body: map[]
2024/06/14 23:49:45 handle command: wake_up (VIN: XP7YGCEK8RB277602)
2024/06/14 23:49:45 Connecting to vehicle...
2024/06/14 23:49:47 sending command "wake_up"...
2024/06/14 23:49:48 The command "wake_up" was successfully executed.
2024/06/14 23:50:06 received command "set_charging_amps" (VIN: XP7YGCEK8RB277602) with body: map[charging_amps:5]
2024/06/14 23:50:07 handle command: set_charging_amps (VIN: XP7YGCEK8RB277602)
2024/06/14 23:50:07 Connecting to vehicle...
2024/06/14 23:50:09 sending command "set_charging_amps"...
2024/06/14 23:50:11 The command "set_charging_amps" was successfully executed.
2024/06/14 23:51:04 received command "{{if .chargeenable}}charge_start{{else}}charge_stop{{end}}" (VIN: XP7YGCEK8RB277602) with body: map[]
2024/06/14 23:51:04 The command "{{if .chargeenable}}charge_start{{else}}charge_stop{{end}}" is not supported.
2024/06/14 23:51:04 received command "wake_up" (VIN: XP7YGCEK8RB277602) with body: map[]
2024/06/14 23:51:05 handle command: wake_up (VIN: XP7YGCEK8RB277602)
2024/06/14 23:51:05 Connecting to vehicle...
2024/06/14 23:51:06 sending command "wake_up"...
2024/06/14 23:51:07 The command "wake_up" was successfully executed.
2024/06/14 23:52:24 received command "{{if .chargeenable}}charge_start{{else}}charge_stop{{end}}" (VIN: XP7YGCEK8RB277602) with body: map[]
2024/06/14 23:52:24 The command "{{if .chargeenable}}charge_start{{else}}charge_stop{{end}}" is not supported.
2024/06/14 23:52:24 received command "wake_up" (VIN: XP7YGCEK8RB277602) with body: map[]
2024/06/14 23:52:25 handle command: wake_up (VIN: XP7YGCEK8RB277602)
2024/06/14 23:52:25 Connecting to vehicle...
2024/06/14 23:52:26 sending command "wake_up"...
2024/06/14 23:52:27 The command "wake_up" was successfully executed.
2024/06/14 23:53:10 received command "{{if .chargeenable}}charge_start{{else}}charge_stop{{end}}" (VIN: XP7YGCEK8RB277602) with body: map[]
2024/06/14 23:53:10 The command "{{if .chargeenable}}charge_start{{else}}charge_stop{{end}}" is not supported.
2024/06/15 00:21:38 received command "set_charging_amps" (VIN: XP7YGCEK8RB277602) with body: map[charging_amps:16]
2024/06/15 00:21:38 received command "{{if .chargeenable}}charge_start{{else}}charge_stop{{end}}" (VIN: XP7YGCEK8RB277602) with body: map[]
[... 171 lines cut ...]
        uri: http://192.168.2.5:5010/currentjson/1
        jq: .charging
      plugged:
[... cut ...]
      uri: "http://192.168.2.110:8080/api/1/vehicles/XP7YGCEK8RB277602/command/{{if .chargeenable}}charge_start{{else}}charge_stop{{end}}"
      method: POST
      body: "" 
    maxcurrent:
      source: http
      uri: http://192.168.2.110:8080/api/1/vehicles/XP7YGCEK8RB277602/command/set_charging_amps
      method: POST
      body: '{"charging_amps": "{{.maxcurrent}}"}'
    wakeup:
      source: http
      uri: http://192.168.2.110:8080/api/1/vehicles/XP7YGCEK8RB277602/command/wake_up
      method: POST
      body: ""

wimaha, 2024-06-15:
> It would be really great, if you could have a look to give me an idea, what I am doing wrong here...

I try!

As you can see in the logs: `[Error]Failed to load private key: open key/private.pem: no such file or directory`:
This means your private.pem was not found. So please check you entered the correct folder in you docker-compose.yml. But it seems you figures out and fixed the problem already, because further down in the logs the error does not appear.

Furthermore your config of evcc is not correct. Since you are using a custom charger, you have to change `"http://192.168.2.110:8080/api/1/vehicles/XP7YGCEK8RB277602/command/{{if .chargeenable}}charge_start{{else}}charge_stop{{end}}"` to `"http://192.168.2.110:8080/api/1/vehicles/XP7YGCEK8RB277602/command/{{if .enable}}charge_start{{else}}charge_stop{{end}}"`

bill340, 2024-06-15:
Thank you sooo much!!
The problem with the private.pem file I figured out myself and it seems to be solved.
I changed the charge command as you mentioned, but still get an error. Here is the log:

2024/06/15 10:37:16 received command "{{if .enable}}charge_start{{else}}charge_stop{{end}}" (VIN: XP7YGCEK8RB277602) with body: map[]
2024/06/15 10:37:16 The command "{{if .enable}}charge_start{{else}}charge_stop{{end}}" is not supported.
2024/06/15 10:37:46 received command "wake_up" (VIN: XP7YGCEK8RB277602) with body: map[]
2024/06/15 10:37:46 handle command: wake_up (VIN: XP7YGCEK8RB277602)
2024/06/15 10:37:46 Connecting to vehicle...
2024/06/15 10:37:47 sending command "wake_up"...
2024/06/15 10:37:48 The command "wake_up" was successfully executed.
2024/06/15 10:38:26 received command "{{if .enable}}charge_start{{else}}charge_stop{{end}}" (VIN: XP7YGCEK8RB277602) with body: map[]
2024/06/15 10:38:26 The command "{{if .enable}}charge_start{{else}}charge_stop{{end}}" is not supported.
2024/06/15 10:38:26 received command "wake_up" (VIN: XP7YGCEK8RB277602) with body: map[]
2024/06/15 10:38:27 handle command: wake_up (VIN: XP7YGCEK8RB277602)
2024/06/15 10:38:27 Connecting to vehicle...
2024/06/15 10:38:28 sending command "wake_up"...
2024/06/15 10:38:29 The command "wake_up" was successfully executed.

pit65535, 2024-06-15:
OK, today I found that the regulation of the charging-Amps is working very well, the only thing that has a Problem ist start charging. As we have enough sunlight now I decided to start Chrging by tesla-app as I guessed it don´t work at all, but then I saw the charging-amps changing in my app.
I now tried to switch off the Charger from evcc-control-panel and it also cannot switch off, it only sets Charging-amps to minimum. Then I set to min+pv
Here ist th according Log:
2024/06/15 12:03:03 sending command "set_charging_amps"...
2024/06/15 12:03:04 The command "set_charging_amps" was successfully executed.
2024/06/15 12:03:41 received command "set_charging_amps" (VIN: 5YJ3E7EB3MF944349) with body: map[charging_amps:14]
2024/06/15 12:03:41 handle command: set_charging_amps (VIN: 5YJ3E7EB3MF944349)
2024/06/15 12:03:41 Connecting to vehicle...
2024/06/15 12:03:43 sending command "set_charging_amps"...
2024/06/15 12:03:44 The command "set_charging_amps" was successfully executed.
2024/06/15 12:04:41 received command "set_charging_amps" (VIN: 5YJ3E7EB3MF944349) with body: map[charging_amps:15]
2024/06/15 12:04:41 handle command: set_charging_amps (VIN: 5YJ3E7EB3MF944349)
2024/06/15 12:04:41 Connecting to vehicle...
2024/06/15 12:04:43 sending command "set_charging_amps"...
2024/06/15 12:04:44 The command "set_charging_amps" was successfully executed.
2024/06/15 12:05:20 received command "set_charging_amps" (VIN: 5YJ3E7EB3MF944349) with body: map[charging_amps:16]
2024/06/15 12:05:20 handle command: set_charging_amps (VIN: 5YJ3E7EB3MF944349)
2024/06/15 12:05:20 Connecting to vehicle...
2024/06/15 12:05:22 sending command "set_charging_amps"...
2024/06/15 12:05:23 The command "set_charging_amps" was successfully executed.
2024/06/15 12:05:41 received command "set_charging_amps" (VIN: 5YJ3E7EB3MF944349) with body: map[charging_amps:15]
2024/06/15 12:05:41 handle command: set_charging_amps (VIN: 5YJ3E7EB3MF944349)
2024/06/15 12:05:41 Connecting to vehicle...
2024/06/15 12:05:43 sending command "set_charging_amps"...
2024/06/15 12:05:44 The command "set_charging_amps" was successfully executed.
2024/06/15 12:06:00 received command "set_charging_amps" (VIN: 5YJ3E7EB3MF944349) with body: map[charging_amps:5]
2024/06/15 12:06:00 handle command: set_charging_amps (VIN: 5YJ3E7EB3MF944349)
2024/06/15 12:06:00 Connecting to vehicle...
2024/06/15 12:06:02 sending command "set_charging_amps"...
2024/06/15 12:06:03 The command "set_charging_amps" was successfully executed.
2024/06/15 12:06:21 received command "set_charging_amps" (VIN: 5YJ3E7EB3MF944349) with body: map[charging_amps:3]
2024/06/15 12:06:22 handle command: set_charging_amps (VIN: 5YJ3E7EB3MF944349)
2024/06/15 12:06:22 Connecting to vehicle...
2024/06/15 12:06:24 sending command "set_charging_amps"...
2024/06/15 12:06:25 The command "set_charging_amps" was successfully executed.
2024/06/15 12:06:40 received command "set_charging_amps" (VIN: 5YJ3E7EB3MF944349) with body: map[charging_amps:1]
2024/06/15 12:06:41 handle command: set_charging_amps (VIN: 5YJ3E7EB3MF944349)
2024/06/15 12:06:41 Connecting to vehicle...
2024/06/15 12:06:43 sending command "set_charging_amps"...
2024/06/15 12:06:44 The command "set_charging_amps" was successfully executed.
2024/06/15 12:06:48 received command "{{if .chargeenable}}charge_start{{else}}charge_stop{{end}}" (VIN: 5YJ3E7EB3MF944349) with body: map[]
2024/06/15 12:06:48 The command "{{if .chargeenable}}charge_start{{else}}charge_stop{{end}}" is not supported.
2024/06/15 12:07:48 received command "set_charging_amps" (VIN: 5YJ3E7EB3MF944349) with body: map[charging_amps:16]
2024/06/15 12:07:49 handle command: set_charging_amps (VIN: 5YJ3E7EB3MF944349)
2024/06/15 12:07:49 Connecting to vehicle...
2024/06/15 12:07:51 sending command "set_charging_amps"...
2024/06/15 12:07:52 The command "set_charging_amps" was successfully executed.

Thank you very much for your help, wimaha, this project is very useful, as tesla always changes thing in their fleetapi......
Maybe you also hav an idea how to solve my last problem :-)

wimaha, 2024-06-15:
Thank you for the logs!
I think it's bug in evcc introduced in version 0.127.0. Wenn you pull version 0.126.6 and use it like `image: evcc/evcc:0.126.6` in the docker-compose.yml it works.
I will create an issue in the evcc repository.

bill340, 2024-06-15:
I saw the issue you created in evcc is closed already and it references to some other issues where it says do not use escaped templates…
But what is the solution for us now? Sorry, but I don’t get it…

bill340, 2024-06-16:
Just installed the new evcc release with your changes implemented...
IT WORKS!!!!
You guys are really awesome!!!
Thanks soo much for your work!!!

closed 2024-06-16 by wimaha