wimaha/TeslaBleHttpProxy #4: Use with more than 1 tesla

bill340, 2024-06-15:
How can I use it with more than one Tesla in regards to the private.pem file?
Do I just run the commands to get a new authentication for the second Tesla? 
But that generates a completely new private.pem file then. What to do with it?

wimaha, 2024-06-15:
You can use the same private.pem for multiple vehicle.

You just habe to repeat the last step from the commands:
`tesla-control -vin YOUR_VIN -ble add-key-request public.pem owner cloud_key`

bill340, 2024-06-15:
The problem is, if I do this, I get the following error:
Error: failed to find a BLE device: can't init hci: no devices available: (hci0: can't down device: device or resource busy)

Any idea? Is it already busy by your Proxy? What could I do?

bill340, 2024-06-15:
Did a 
docker-compose down
That solved the problem.

closed 2024-06-15 by wimaha