wimaha/TeslaBleHttpProxy #5: DISCUSSION - Full EVCC config

MrBLJ, 2024-06-15:
>[!important]
> This is not an issue

Could you please provide your full EVCC config ? (discussions are not enabled, sorry for posting here)

infinus, 2024-06-15:
Setup: EVCC and MQTT running on a raspberry pi. You will need to use Mqtt or some other integration to be able to pull the SOC of your car to make it work. Until I did this I could not get it to run out of EVCC. Since I already had a teslamate instance running I used that.

```
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: info

# unique installation id
plant: *******

interval: 10s # control cycle interval

sponsortoken: *******

meters:
- type: template
  template: enphase
  usage: grid
  host: 192.168.1.225
  token: ******
  name: grid1

- type: template
  template: enphase
  usage: pv
  host: 192.168.1.225
  token: *****

chargers:
- type: template
  template: twc3
  host: 192.168.1.108
  name: TWC

vehicles:
- name: Infinus
  type: custom
  capacity: 74
  chargeenable:
    source: http
    uri: "http://192.168.1.17:8080/api/1/vehicles/*****/command/{{if .chargeenable}}charge_start{{else}}charge_stop{{end}}"
    method: POST
    body: ""
  maxcurrent: # set charger max current (A)
    source: http
    uri: http://192.168.1.17:8080/api/1/vehicles/*****/command/set_charging_amps
    method: POST
    body: '{"charging_amps": "{{.maxcurrent}}"}'
  wakeup: # vehicle wake up command
    source: http
    uri: http://192.168.1.17:8080/api/1/vehicles/*****/command/wake_up
    method: POST
    body: ""
  soc:
    source: mqtt
    topic: teslamate/cars/1/battery_level
    timeout: 60s
  range:
    source: mqtt
    topic: teslamate/cars/1/est_battery_range_km
    timeout: 60s
  status:
    source: combined
    plugged:
      source: mqtt
      topic: teslamate/cars/1/plugged_in
      timeout: 60s
    charging:
      source: mqtt
      topic: teslamate/cars/1/charging_state
      timeout: 60s

loadpoints:
- title: Home
  charger: TWC
  vehicle: Infinus
  mode: pv

mqtt:
  broker: localhost:1883

tariffs:
  currency: USD
  grid:
    # static grid price summer
    type: fixed
    price: 0.0925875 # [currency]/kWh
    zones:
      - days: Mo-Fr
        hours: 16-19
        price: 0.25797 # USD/kWh
      - days: Sa,So
        price: 0.0925875 # USD/kWh
    # Static grid price winter:
#    type: fixed
#    price: 0.0926325 # [currency]/kWh
#    zones:
#      - days: Mo-Fr
#        hours: 10-15
#        price: 0.0262125 # USD/kWh
#        hours: 16-19
#        price: 0.2440725 # USD/kWh
#      - days: Sa,So
#        price: 0.0926325 # USD/kWh

site:
  title: Home
  meters:
    grid: grid1
    pv:
    - pv1

```

MrBLJ, 2024-06-15:
Thanks a lot this is helpful. I'm trying to figure out how to setup the system In my case : I have a [my dumb wallbox](https://hager.com/intl-en/solutions/energy-distribution/witty-charging-solutions) for one car (model 3), and a simple plug for the other 'model y). I'm not sure about how to overcome the _charger_ part of the config.

The full config you posted will help me get a better understanding of the system.

infinus, 2024-06-15:
Honestly EVCC is great but it would be awesome if they could make a Tesla specific implementation that does not rely on the wall charger at all but instead identified if it is home or not based on location and received the connected status from the vehicle and not the charger.

MrBLJ, 2024-06-15:
[Looking at this ](https://github.com/speedst3r/twc3teslamate) it appears possible. Setup is getting complex though, i'll need to have a better look at it during the week.

wimaha, 2024-06-15:
@MrBLJ Are you using a custom charger in your config?

MrBLJ, 2024-06-16:
> @MrBLJ Are you using a custom charger in your config?

For now I’m exploring different options. Evcc is not in use yet. Would you recommend using a custom charger ?

wimaha, 2024-07-07:
Since version 0.128.0 or newer of evcc it is very easy to integrate ble proxy! I’ve updated the readme.

closed 2024-07-07 by wimaha