wimaha/TeslaBleHttpProxy #8: Expand the documentation

Likemike24, 2024-06-18:
Could you please expand the documentation for dummies? I´m not really firm with GO and Docker, so it would be nice to get the instructions to install and set the required components for using BLE-Proxy on a Raspberry Pi.

wimaha, 2024-06-24:
I will update the documentation this week so that a detailed guide on how to handle these values will be available.

dglaude, 2024-06-24:
> I will update the documentation this week so that a detailed guide on how to handle these values will be available.

Thank you very much in advance, any improvement is appreciated.

I am circulating the various issues/discussions and the various solutions.

There are so many possibilities and steps, such as install evcc, teslamate, this ble proxy, and what else. Also you have to understand docker, maybe install golang and compile your first program. And when you have all the software pieces, you have to configure evcc the proper way. This is fun but really a lot (depending on your starting point).

Right now, I have strong hope in the following command to build locally the container on my machine (but I am not even sure that is the right way): ```docker build -t wimaha/tesla-ble-http-proxy https://github.com/wimaha/TeslaBleHttpProxy.git```

svwhisper, 2024-06-26:
The additional doco would be very helpful.  I finally got mine working with Node Red, though I gave up on docker.  Now I feel somewhat insulated from what Tesla does with their new charging-for-the-api regime. 

Thanks for your effort on this little project.

wimaha, 2024-06-27:
I have added some more instructions here: https://github.com/wimaha/TeslaBleHttpProxy/blob/main/docs/installation.md

alexonbandit, 2024-06-28:
Hi @wimaha, when following the installation procedure I am getting issues at the built process.
In STEP2 after I did the clone github I receive the following error:

alex@raspberrypizero:~/vehicle-command $ go get ./...
warning: GOPATH set to GOROOT (/home/alex/.local/share/go) has no effect
go: downloading github.com/google/shlex v0.0.0-20191202100458-e7afc7fbc510
...
...

alex@raspberrypizero:~/vehicle-command $ go build ./...
warning: GOPATH set to GOROOT (/home/alex/.local/share/go) has no effect
# runtime/cgo
gcc: error: unrecognized command-line option '-marm'

Do you know what went wrong? Many thanks in advance!

Likemike24, 2024-06-28:
> Hi @wimaha, when following the installation procedure I am getting issues at the built process. In STEP2 after I did the clone github I receive the following error: [...]

You have to use the 32 bit version of the OS and the correct version of Go.

alexonbandit, 2024-06-28:
Many thanks! 
True I installed the 64 bit version!
I will give it a try on the weekend.

wimaha, 2024-06-28:
I've changed the description to the 64 bit version.

dglaude, 2024-07-03:
Oh I had that '-marm' issue on my Pi4 and I figured that I was running the 64 bit version... since I had a NUC nearby I gave up on the Pi installation. And it goes much faster on NUC (the performance might not be needed for running the software, but to compile and create the container... be ready to wait).

However, I found a neat way to compile the software with go avoiding most of the problem by compiling in a container. So that could be one alternative option. And I have also seen "two stage" docker file trick...

Just remember that at the end, I did not do it on my Pi... but I am still interested because maybe one day I will run this on a PiZero 2W or something with that kind of size.

wimaha, 2024-07-07:
I will add a feature, so you can generate the needed keys and send them to the vehicle without installing extra software. If you want, you can test it: https://github.com/wimaha/TeslaBleHttpProxy/pull/23

wimaha, 2024-07-08:
The new way of generating the keys and sending to the vehicle without additional software is published. So the problems above with generating the keys with vehicle-command are solved. I think we can close this issue. :-)

closed 2024-07-08 by wimaha