wimaha/TeslaBleHttpProxy #17: set_charging_amps failing from evcc with tesla template with commandProxy

FraBoCH, 2024-06-30:
Starting testing https://github.com/evcc-io/evcc/pull/14616 with the docker image provided by evcc maintainer (https://hub.docker.com/layers/evcc/evcc/tesla/images/sha256-7e6d75f3a55a8c02076bf0745dbfd0f698f434258cdc4c59ee6ce38a5ea0fb32?context=explore): 

Command set_charging_amps is not coming through:
2024/06/30 12:54:53 ERRO The command was canceled. command=set_charging_amps err="charing Amps missing in body"

As I understand it @wimaha, this seem to be an issue of this proxy as evcc is just sending the same requests but to your proxy instead of evcc one, right ?

wimaha, 2024-06-30:
What is the body in the logs? It should be printed in the line with `received command`.

FraBoCH, 2024-06-30:
Ok, here is the trace log from evcc: 
```
[tesla ] TRACE 2024/06/30 15:00:32 POST http://192.168.X.Y:8080/api/1/vehicles/YYYYYYYY/command/set_charging_amps
[tesla ] TRACE 2024/06/30 15:00:32 {"charging_amps": 1}
--
{"response":{"result":true,"reason":"The command was successfully received and will be processed shortly.","vin":"YYYYYY","command":"set_charging_amps"}}
[lp-1  ] ERROR 2024/06/30 15:00:32 max charge current 1A: 201 Created
``` 

From what I understand, the body is thus in JSON: `{"charging_amps": 1}`, which by the way is consistent with the docs https://developer.tesla.com/docs/fleet-api#set_charging_amps.

FraBoCH, 2024-06-30:
So I think the issue is that your code is expecting a string and not directly an for charging_amps !
In other words, it works when sending `{"charging_amps": "1"} ` but fails with `{"charging_amps": 1} `.

wimaha, 2024-06-30:
In the docs is stated `"charging_amps": "integer"`

So I think expecting a string is correct, because integer is here integrated in a string.

Anyway I will make an update so that integer are also accepted.

FraBoCH, 2024-06-30:
Seems cleaner anyway to sent a « pure » int instead of wrapping it up inside a string.

wimaha, 2024-06-30:
Fixed with 1.0.4 🙂

closed 2024-06-30 by wimaha

FraBoCH, 2024-06-30:
Nice! I can confirm that the set_charging_amp command from EVCC is now properly executed!