wimaha/TeslaBleHttpProxy #19: Returning HTTP status 201 Created is considered as an error by EVCC tesla client

FraBoCH, 2024-06-30:
I-m experiencing another issue when using this proxy with evcc standard tesla client (see ).

It looks like the proxy is returning HTTP status 201 created to every command which is considered as an error by EVCC which logs are thus filled with errors like `[lp-1 ] ERROR 2024/06/30 22:21:37 max charge current 1A: 201 Created`.

Strangely enough, this doesn’t happen when using a custom vehicle.

Anyway, I strongly suggest simply returning the classic http statut « 200 OK ».

Waiting for a new release to continue testing ;-) Looks like we are almost there!

wimaha, 2024-06-30:
ok ... fully correct would be http status 202 😉
For now I changed the status code to 200 to match the Tesla API. But I think the implementation in evcc is not correct. Instead of the http status code the bool value in `Result` should be used.

Fixed with 1.0.5 (https://github.com/wimaha/TeslaBleHttpProxy/pull/20)

closed 2024-06-30 by wimaha

FraBoCH, 2024-07-01:
I agree with 202, but I will let you play to see if evcc like it too or not ;-)
Anyway I confirm everything now works flawlessly with evcc nightly now. Nice !