wimaha/TeslaBleHttpProxy #22: charging amps goes to zero

wimaha, 2024-07-07:
Interesting behaviour...
I notice that when the proxy throws this error, my car goes to zero amps (as shown on the app and in my energy monitor) and then quickly climbs back up to the target.  When I kill the proxy, the odd behaviour goes away.

The odd behaviour doesn't happen every time I see this error, but whenever actual charge rate goes to zero, the error was thrown seconds prior.

Thoughts?

2024/07/07 12:31:46 INFO TeslaBleHttpProxy is loading ...
2024/07/07 12:31:46 DEBU LogLevel set to debug
2024/07/07 12:31:46 INFO TeslaBleHttpProxy is running!
2024/07/07 12:32:04 INFO received command command=set_charging_amps VIN=LRWYHCEL4PC726XYZ body=map[charging_amps:23]
2024/07/07 12:32:05 INFO handle command command=set_charging_amps VIN=LRWYHCEL4PC726XYZ
2024/07/07 12:32:06 WARN failed to connect to vehicle (A): ble: failed to enumerate device services: ATT request failed: input channel closed: io: read/write on closed pipe
2024/07/07 12:32:06 INFO retrying in 3 seconds
2024/07/07 12:32:10 WARN failed to connect to vehicle (A): ble: failed to enumerate device services: ATT request failed: input channel closed: io: read/write on closed pipe
2024/07/07 12:32:10 INFO retrying in 6 seconds
2024/07/07 12:32:19 INFO The command was successfully executed. command=set_charging_amps

_Originally posted by @svwhisper in https://github.com/wimaha/TeslaBleHttpProxy/issues/6#issuecomment-2212294545_

wimaha, 2024-07-07:
@svwhisper
Opened a new issue for this, as it is not related to the other case.
Is 23 A a correct value? I think the model y can charge with up to 16 A. Maybe the car sets charging amps to zero when incorrect values are received.

CrazyMgix, 2024-07-07:
The Model Y can charge up to 32A with one phase

svwhisper, 2024-07-07:
Yes, up to 32A single phase 240v.

svwhisper, 2024-07-08:
Two thoughts, based on zero evidence:

1. The proxy sends either a malformed set_charging_amps request or an erroneous one with 0 amps
2. The rate at which I'm requesting set_charging_amps amps is too great-- once every 20 seconds-- and this is exposing a defect in either the proxy or the car's firmware

I can't charge today as there's too little solar and I'm off-grid.  I'll vary the rate at which the request is sent to see if that makes any difference.

wimaha, 2024-07-08:
You can review the code. I don't see any possibility that a malformed request ist send. I will add additional logs for that case tomorrow, so we can see what parsed amp values are send.

svwhisper, 2024-07-11:
I haven't seen this happen again.  Please close this one.  thanks

closed 2024-07-11 by wimaha