wimaha/TeslaBleHttpProxy #28: How close does the BLE device has to be

tiehfood, 2024-07-14:
Hi,

I'm using a raspberry pi zero as BLE device.
Is there a maximum distance between the BLE device and the vehicle in order to work properly?

tiehfood, 2024-07-14:
There also seems to be a "vehicle (B)" some times. Does TeslaBlehttpProxy remember old vehicles?
I had so setup the key twice initially, so is there the "old" car cached somehow?

wimaha, 2024-07-17:
`vehicle (B)` does not refer to a second vehicle. Is it just a stage when connecting to the car.

So you have to move the BLE device closer to the car. In my case, I'am using a raspberry which is about 3 meters away from the car.

tiehfood, 2024-07-21:
Ok, I bought an external bluetooth adapter and parked closer to the raspberry - about 1m with a wall between them. I still have problems connecting to it...
[image]

dglaude, 2024-07-24:
TL;DR:
* I think there is a problem somewhere in the software with its current version ( see #6 )
* I don't think you need to be very close, but the closer the better
* you can test with `tesla-control`
* careful if you have two Bluetooth interface

Right now, there is something currently a bit random with the software... some days it is working perfectly fine, and days no message are exchange.

At one point (when it was not working) I decided to moved closer, and suddenly it started to work, but I did not knew if it was the distance (or the shutdown, or the cooldown of the equipment or ???). The problem is that today it stopped working again, and no restart change a thing.

I do know it is not an issue with the distance, because it did work from a bigger distance, and it is manually working with tesla-control`.

Notice that on my NUC, at one point the hci0 did vanish, and the Bluetooth device reappeared as hci1. However, both TeslaBleHttpProxy and `tesla-control` were unable to work (for the later, there are issues complaining about the fact that it does not work with hci1).

Now, if you added an external BLE dongle, it might be visible as hci1 and your current build-in Bluetooth is still hci0... so maybe you are not using the BLE interface you think you should be using.

tiehfood, 2024-07-25:
I disabled the build in bluetooth device via the boot settings. `hcitool` tool only shows the external USB dongle, so this should be fine. I now moved the antenna again closer to the car and now it seems to work. At least most of the time. As you said, there seems to be some kind of randomness in the system from time to time.

Arturfrain, 2024-08-21:
I have the same issue. Initially I was setting up my PI4 and it was working. The Pi is approx 4-5 m from the car. Today i tried to charge with Evcc but nothing happened. The “docker logs --since 12h tesla-ble-http-prox” says “failed to connect to vehicle (A)….”
[image]

wimaha, 2024-08-22:
#6 is resolved with version 1.2.4

For the other problems I can tell you that this program uses the vehicle-command library. So the code which connects to the vehicle is the same as `tesla-control`. The proxy is for some use case just more convenient.

wimaha, 2024-09-08:
I'm closing, because I think the problem is solved.

closed 2024-09-08 by wimaha