wimaha/TeslaBleHttpProxy #29: Key name in dashboard? Using two instances for the same car?

dglaude, 2024-07-15:
This is more a support question.

I have one working setup on a first Pi, but I try to setup a new Pi to have all the software component on a single machine.

However, doing the key setup with /dashboard on the second Pi failed with `Error: failed to connect to vehicle (A): context deadline exceeded` (worked fine on the first Pi). I wonder if it could be related to a "friendly key name" and since you cannot specify it, it would be the same for both attempt?

If such a thing as a "key name", then it should be presented in the dashboard with a good default name.

If everything else fail, maybe I will just try to reuse the key from the other Pi.

dglaude, 2024-07-17:
This whole issue was based on the wrong assumption that either `owner` or `cloud_key` from the following line was the name of the key:
`tesla-control -vin YOUR_VIN -ble add-key-request public.pem owner cloud_key`

Because nothing was working anymore, I tried on my NUC (creating the container or running TeslaBleHttpProxy => fail again) then I had success with `tesla-control` and that is where I figured that nor `owner` nor `cloud_key` could be changed... well I was considering if `charging_manager` could be appropriate in place of `owner`.

Now my Pi is working again, and I don't have a clue what changed.

Maybe this issue could be closed now, or there is something to discuss, like way to detect communication problem and trigger an alarm.

wimaha, 2024-07-17:
You can use multiple cars with the same key!
It is not possible to change the name of the key. They named "unknown" or something like this in the car. To rename the key you must use the fleet api.

`Error: failed to connect to vehicle (A): context deadline exceeded` This error could be that your car was sleeping. You have to manually wake the car during the key setup.

dglaude, 2024-07-21:
Thanks for the response, my problem seems to be BLE/Wakeup related, very similar to what is experienced in #6.
The issue could be something related to the car and waking up. I'll keep reading and investigate.

closed 2024-07-21 by dglaude