wimaha/TeslaBleHttpProxy #34: No NFC confirmation request

faxioman, 2024-07-20:
Hi,
I'm trying to associate my Model 3 2024:

```
2024/07/20 18:45:23 INFO Key-Request connection established
2024/07/20 18:45:23 INFO sending command=add-key-request body=map[]
2024/07/20 18:45:23 INFO Sent add-key request to LRW3*************8. Confirm by tapping NFC card on center console.
2024/07/20 18:45:23 INFO successfully executed command=add-key-request body=map[]
```

But nothing happens in my car.
I'm using a Mac Book pro with bluetooth connected to my car.
I'm sure I'm missing somehting... 😅

wimaha, 2024-07-20:
What happens when you confirm by tapping NFC card on center console?

faxioman, 2024-07-20:
The problem is that no confirmation request appears on my car

wimaha, 2024-07-20:
This is expected behavior! You have to confirm by tapping NFC card on center console. Then a message will appear on screen.

faxioman, 2024-07-20:
I'll try ... now! 😉

faxioman, 2024-07-20:
Ok, now I can see the key!! Thanks!
At this moment, all the APIs request get a 404 error ... but this is another thing to check. Thanks!!

closed 2024-07-20 by faxioman

phoet, 2024-10-07:
The send_keys methd hangs indefinitively for me when i submit the vin. There is not much in the logs

```
2024/10/07 10:15:15 INFO ECDSA private key generated and saved
2024/10/07 10:15:15 INFO ECDSA public key generated and saved
2024/10/07 10:15:15 DEBU privateKeyFile loaded
2024/10/07 10:15:15 INFO BleControl initialized
2024/10/07 10:15:24 DEBU connecting to vehicle (A)...
2024-10-07T10:15:24+02:00 [debug] Creating new BLE device
2024-10-07T10:15:24+02:00 [debug] Searching for BLE beacon S43172b7a80223604C...
```

Since I try this from my mac, I attached it to this post where it looks like it should work on one.