wimaha/TeslaBleHttpProxy #39: Issue Pairing Tesla Y

JohnnyFFM, 2024-08-01:
Hi, 

I've managed to install TeslaBleHttpProxy on a Pi Zero W but struggling with the initial setup.

I've created keys, sent them to the vehicle and confirmed by putting an NFC card on center console (on the phone charger). There was no visual confirmation, just the logs show a success. I then tried to make the lights flash to test the setup, but didnt work. 

Flash Lights test:
```
curl -X POST http://192.168.1.49:8080/api/1/vehicles/XP123456789/command/flash_lights
{"response":{"result":true,"reason":"The command was successfully received and will be processed shortly.","vin":"XP123456789","command":"flash_lights"}}
```

Log:
```
Aug 01 21:07:11 teslapi TeslaBleHttpProxy[387]: 2024/08/01 21:07:11 INFO ECDSA public key generated and saved
Aug 01 21:07:11 teslapi TeslaBleHttpProxy[387]: 2024/08/01 21:07:11 INFO BleControl initialized
Aug 01 21:07:36 teslapi TeslaBleHttpProxy[387]: 2024/08/01 21:07:36 INFO Key-Request connection established
Aug 01 21:07:36 teslapi TeslaBleHttpProxy[387]: 2024/08/01 21:07:36 INFO sending command=add-key-request body=map[]
Aug 01 21:07:36 teslapi TeslaBleHttpProxy[387]: 2024/08/01 21:07:36 INFO Sent add-key request to XP123456789. Confirm by tapping NFC card on center console.
Aug 01 21:07:36 teslapi TeslaBleHttpProxy[387]: 2024/08/01 21:07:36 INFO successfully executed command=add-key-request body=map[]
Aug 01 21:08:17 teslapi TeslaBleHttpProxy[387]: 2024/08/01 21:08:17 INFO received command=flash_lights body=map[]
Aug 01 21:08:17 teslapi TeslaBleHttpProxy[387]: 2024/08/01 21:08:17 INFO connecting to Vehicle ...
Aug 01 21:08:18 teslapi TeslaBleHttpProxy[387]: 2024/08/01 21:08:18 WARN failed to perform handshake with vehicle (A): vehicle rejected request: your public key has not been paired with the vehicle
Aug 01 21:08:18 teslapi TeslaBleHttpProxy[387]: 2024/08/01 21:08:18 INFO retrying in 3 seconds
Aug 01 21:08:23 teslapi TeslaBleHttpProxy[387]: 2024/08/01 21:08:23 WARN failed to perform handshake with vehicle (A): vehicle rejected request: your public key has not been paired with the vehicle
Aug 01 21:08:23 teslapi TeslaBleHttpProxy[387]: 2024/08/01 21:08:23 INFO retrying in 6 seconds
Aug 01 21:08:30 teslapi TeslaBleHttpProxy[387]: 2024/08/01 21:08:30 ERRO stop retrying after 3 attempts error="failed to perform handshake with vehicle (A): vehicle rejected request: your public key has not been paired with the vehicle"
```

Any hint on what I'm missing?

Thanks, 

JohnnyFFM

wimaha, 2024-08-01:
Hi,

in the logs is clearly stated why the command is not working: `your public key has not been paired with the vehicle`.

After `Sent add-key request to VIN. Confirm by tapping NFC card on center console.`and you are tapping the NFC card on center console, you get a message on the vehicle screen.

JohnnyFFM, 2024-08-01:
I finally found my mistake. "tapping on center console" was the issue. the exact spot to tap seems to be between armrest and cup holders. i was tapping around the phone chargers 🤦 Now working like a charm!

Thx for your help and the code! 

JohnnyFFM

closed 2024-08-01 by JohnnyFFM