wimaha/TeslaBleHttpProxy #42: Can’t open dashboard

Arturfrain, 2024-08-06:
Hi, I try to install the BLE proxy on a PI Zero. I succeeded in with the image followed by dev.
After docker compose up -d
It says 
[+] running 1/0
Container Tesla-ble-http-proxy  Started.

now as in the documentation said I want to browse to http://“my-pizero-ip”:8080/dashboard
But no connection can be established. 
I have no clue what I do wrong. Maybe is it a problem with the pi zero? Do I need an other Raspberry Pi?

I do this test away from the car. Does the dashboard need me to be in the near of the car to appear?

Any help is appreciated

wimaha, 2024-08-10:
Can you please restart the container and copy the logs (`docker logs --since 12h tesla-ble-http-proxy`). Thank you!

Arturfrain, 2024-08-11:
Thank you, I am away from home until Wednesday. I will do that then and post.

Arturfrain, 2024-08-16:
Hallo, ich habe nun auf einem PI4 unter 64bit intstalliert. Es funktioniert alles.  Ich habe evcc mit dem Sponsortoken un dem Tesla Template laufen. Die Http Proxy arbeitet sehr schnell. Man verwendet ja nach wie vor die Tesla Fleet API zur Abfrage des SOC. Gibt es dadurch eventuelle Einschränkungen bei der Steuerung im Modus PV ?
Danke für das tolle Programm !!

closed 2024-08-19 by Arturfrain