wimaha/TeslaBleHttpProxy #49: Set Charging amps no more working

pit65535, 2024-08-22:
I use tesla ble Proxy for charging with evcc since tesla restricted the number of calls to the fleet-api. Since then it worked perfect, but today I saw the charging amps are not changing anymore (I guess it´s after the Update of the Tesla Software, it is now: v12 (2024.26.7)).

the Log:
docker logs --since=1h tesla-ble-http-proxy
2024/08/22 12:49:21 INFO TeslaBleHttpProxy 1.2.3 is loading ...
2024/08/22 12:49:21 INFO BleControl initialized
2024/08/22 12:49:21 INFO TeslaBleHttpProxy is running!
2024/08/22 12:50:01 INFO received command=set_charging_amps body=map[charging_amps:13]
2024/08/22 12:50:01 INFO connecting to Vehicle ...
2024/08/22 12:50:01 INFO received command=charge_start body=map[]
2024/08/22 12:50:17 WARN failed to perform handshake with vehicle (B): context deadline exceeded
2024/08/22 12:50:17 INFO retrying in 3 seconds
2024/08/22 12:50:22 can't accept: listner timed out
2024/08/22 12:50:22 INFO connection established
2024/08/22 12:50:22 INFO sending command=set_charging_amps body=map[charging_amps:13]
2024/08/22 12:50:23 WARN failed to set charging Amps to 13: car could not execute command: SetChargingAmps failed
2024/08/22 12:50:23 INFO retrying in 3 seconds
2024/08/22 12:50:28 WARN failed to set charging Amps to 13: car could not execute command: SetChargingAmps failed
2024/08/22 12:50:28 INFO retrying in 6 seconds
2024/08/22 12:50:34 ERRO canceled command=set_charging_amps body=map[charging_amps:13] err="failed to set charging Amps to 13: context deadline exceeded"
2024/08/22 12:50:35 INFO connecting to Vehicle ...
2024/08/22 12:50:38 INFO connection established
2024/08/22 12:50:38 INFO sending command=charge_start body=map[]
2024/08/22 12:50:39 INFO successfully executed command=charge_start body=map[]
2024/08/22 12:50:54 INFO received command=set_charging_amps body=map[charging_amps:16]
2024/08/22 12:50:54 INFO sending command=set_charging_amps body=map[charging_amps:16]
2024/08/22 12:50:55 WARN failed to set charging Amps to 16: car could not execute command: SetChargingAmps failed
2024/08/22 12:50:55 INFO retrying in 3 seconds
2024/08/22 12:50:59 WARN failed to set charging Amps to 16: car could not execute command: SetChargingAmps failed
2024/08/22 12:50:59 INFO retrying in 6 seconds

I am not sure, if it´s a bug of teslaBleProxy or of the new tesla Software.....

pit65535, 2024-08-22:
I rebooted the vehicle, after that everything works fine again.

closed 2024-08-22 by pit65535

wimaha, 2024-08-22:
Thank you for sharing.