wimaha/TeslaBleHttpProxy #63: Make vehicle commands optionally block

michael-robbins-seek, 2024-12-13:
I can see in the vehicle data handling you've got the ability to block in there for the response.

Could we possibly make it so the vehicle commands code can also optionally block too for the command to be executed? (possibly including if it was successful or not?)

Use case is in my HA automation, I'd like to:
1. Set charging amps
2. Start car charging
3. Get vehicle data confirming the current charging amps and that charging is started

After 1, I can add in a validation step to see that we did indeed set charging amps successfully
After 2, I can add in a validation step to see that we did indeed start charging
Before 3, I can then confidently run the 'get vehicle data' step knowing we've blocked and the above command have run

michael-robbins, 2024-12-15:
Another scenario here maybe could be to tell if the car is asleep maybe? Just trying to think through the scenarios where I want to occasionally poll the car if it's awake (eg charging). There's other ways I can tell if the cars charging (via the wall charger).

Let me try out a few things when the cars asleep, I wonder if over the bluetooth connection we can tell the cars there, just asleep, vs, the car *isn't there* 🤔

sftman18, 2024-12-18:
I have a similar situation where confirmation of the completion of the command would be helpful.  With my previous direct usage of the tesla-control ble library, I would get a failure response condition through "context deadline exceeded" or "read/write on closed pipe" through stderr, which I could respond to with an immediate retry.
If I could confirm that the command was successfully completed (vs fire and forget), that would help with my state determination.

@michael-robbins In terms of knowing if the car is awake, without waking it, there is a command available in the tesla-control ble library.
```
tesla-control -ble <...> body-controller-state
 {
 	"closureStatuses": {
 		"frontDriverDoor": "CLOSURESTATE_CLOSED",
 		"frontPassengerDoor": "CLOSURESTATE_CLOSED",
 		"rearDriverDoor": "CLOSURESTATE_CLOSED",
 		"rearPassengerDoor": "CLOSURESTATE_CLOSED",
 		"rearTrunk": "CLOSURESTATE_CLOSED",
 		"frontTrunk": "CLOSURESTATE_CLOSED",
 		"chargePort": "CLOSURESTATE_OPEN",				Open means a cable is attached!		Only available when car is AWAKE. Need to wake before this is reliability read
 		"tonneau": "CLOSURESTATE_CLOSED"
 	},
 	"vehicleLockState": "VEHICLELOCKSTATE_UNLOCKED",
 	"vehicleSleepStatus": "VEHICLE_SLEEP_STATUS_AWAKE",		Available when car is asleep. Can directly read if car is awake
 	"userPresence": "VEHICLE_USER_PRESENCE_NOT_PRESENT"
 }
```
vehicleSleepStatus, has these states:
```
 VEHICLE_SLEEP_STATUS_UNKNOWN
 VEHICLE_SLEEP_STATUS_AWAKE 
 VEHICLE_SLEEP_STATUS_ASLEEP
```

wimaha, 2024-12-18:
I will add the Body Controller State via seperate API-Call `/api/1/vehicles/{vin}/body_controller_state`. 

You can test this and the "Wait Option" on dev. (No Docker Build yet) If you want to wait until a command es executed and get a feedback you can add `wait=true`. Example: `http://localhost:8080/api/1/vehicles/VIN/command/set_charge_limit?wait=true`.

michael-robbins, 2024-12-19:
`body_controller_state` would be amazing!
And `wait=true` is awesome!

Do you have a way to accept donations or anything? This stuff is epic!!

michael-robbins, 2024-12-21:
@wimaha with the `wait=true` code in `dev` atm, wouldn't https://github.com/wimaha/TeslaBleHttpProxy/blob/dev/main.go#L145 mean the commands are run twice if `wait=true`? Should that section be in an `else` statement?

wimaha, 2024-12-21:
@michael-robbins: There is es `return` above, so no it is not executed twice. :-)

wimaha, 2024-12-21:
Just published version 2.0.0 with the features integrated! :-)

Please note, that I wrapped the response in the `response struct` to be consistent with the command api. So you have to make a small change to your integration.

closed 2024-12-21 by wimaha