wimaha/TeslaBleHttpProxy #65: unable to get vehicle SOC using TeslaBLE Proxy + EVCC

silencer2008, 2024-12-18:
Unable to retrieve vehicle SOC.
[image]

vehicles:
  - name: Tesla
    type: custom
    title: Blue Beetle
    icon: car
    capacity: 60
    chargeenable:
      source: http
      uri: "http://192.168.4.150:8080/api/1/vehicles/VIN/command/{{if .chargeenable}}charge_start{{else}}charge_stop{{end}}"
      method: POST
      body: ""
    maxcurrent: # set charger max current (A)
      source: http
      uri: http://192.168.4.150:8080/api/1/vehicles/VIN/command/set_charging_amps
      method: POST
      body: '{"charging_amps": "{{.maxcurrent}}"}'
    wakeup: # vehicle wake up command
      source: http
      uri: http://192.168.4.150:8080/api/1/vehicles/VIN/command/wake_up
      method: POST
      body: ""
    soc:
      source: http
      uri: http://192.168.4.150:8080/api/1/vehicles/VIN/vehicle_data?endpoints=charge_state
      method: GET
      jq: .response.charge_state.battery_level
      timeout: 30s 
    limitsoc:
      source: http
      uri: http://192.168.4.150:8080/api/1/vehicles/VIN/vehicle_data?endpoints=charge_state
      method: GET
      jq: .response.charge_state.charge_limit_soc
      timeout: 30s
    range:
      source: http
      uri: http://192.168.4.150:8080/api/1/vehicles/VIN/vehicle_data?endpoints=charge_state
      method: GET
      jq: .response.charge_state.battery_range
      scale: 1.60934
      timeout: 30s

Is this expected that I should use the Tesla API for SOC ? Stop and Start works fine. Just error on SOC

wimaha, 2024-12-18:
Are you updated to the new version?

silencer2008, 2024-12-18:
Yes, up to date. First thing i checked

wimaha, 2024-12-18:
Can you please post the logs from the proxy?

silencer2008, 2024-12-18:
How do I access proxy logs ?

wimaha, 2024-12-18:
How did you check the version? 

If you installed via docker you can use this command:
`docker logs --since 12h tesla-ble-http-proxy`

silencer2008, 2024-12-18:
I did this to check version based on the Readme
docker pull wimaha/tesla-ble-http-proxy
docker compose up -d

Using default tag: latest
latest: Pulling from wimaha/tesla-ble-http-proxy
Digest: sha256:94518e8774b9ff25037e35e10aadac7ea9a48016dba0e7841d91dc4130915dd6
Status: Image is up to date for wimaha/tesla-ble-http-proxy:latest
docker.io/wimaha/tesla-ble-http-proxy:latest

Logs for proxy show this when error above occurs: 
2024/12/18 13:23:42 INFO connecting to Vehicle ...
2024/12/18 13:23:46 INFO connection established
2024/12/18 13:23:46 INFO sending command=set_charging_amps body=map[charging_amps:5]
2024/12/18 13:23:46 INFO successfully executed command=set_charging_amps body=map[charging_amps:5]
2024/12/18 15:40:23 INFO received command=charge_stop body=map[]
2024/12/18 15:40:23 INFO connecting to Vehicle ...
2024/12/18 15:40:26 INFO connection established
2024/12/18 15:40:26 INFO sending command=charge_stop body=map[]
2024/12/18 15:40:27 INFO successfully executed command=charge_stop body=map[

Looks like no info regarding Get data ?

silencer2008, 2024-12-18:
Ok, may have fixed the issue by manually updating the container. For some reason it was still running 1.2.7.
Can see the Vehicle Data being pulled by the proxy now. 

2024/12/18 17:00:28 INFO connecting to Vehicle ...
2024/12/18 17:00:31 INFO connection established
2024/12/18 17:00:31 INFO sending command=vehicle_data body=map[endpoints:[charge_state]]
2024/12/18 17:00:33 INFO successfully executed command=vehicle_data body=map[endpoints:[charge_state]]
2024/12/18 17:00:33 INFO sending command=vehicle_data body=map[endpoints:[charge_state]]
2024/12/18 17:00:34 INFO successfully executed command=vehicle_data body=map[endpoints:[charge_state]]
2024/12/18 17:04:16 INFO connecting to Vehicle ...
2024/12/18 17:04:19 INFO connection established
2024/12/18 17:04:19 INFO sending command=vehicle_data body=map[endpoints:[charge_state]]
2024/12/18 17:04:20 INFO successfully executed command=vehicle_data body=map[endpoints:[charge_state]]
2024/12/18 17:04:20 INFO sending command=vehicle_data body=map[endpoints:[charge_state]]
2024/12/18 17:04:21 INFO successfully executed command=vehicle_data body=map[endpoints:[charge_state]]
2024/12/18 17:04:21 INFO sending command=vehicle_data body=map[endpoints:[charge_state]]
2024/12/18 17:04:23 INFO successfully executed command=vehicle_data body=map[endpoints:[charge_state]]
2024/12/18 17:04:23 INFO received command=charge_stop body=map[]
2024/12/18 17:04:23 INFO sending command=charge_stop body=map[]
2024/12/18 17:04:24 INFO successfully executed command=charge_stop body=map[]

Any idea as to why the container didnt update as expected ?

wimaha, 2024-12-18:
Maybe caching issue. Nice that it works now! 😊

closed 2024-12-18 by wimaha