wimaha/TeslaBleHttpProxy #76: TeslaBleHttpProxy gets into a bad state, eats memory and cpu

skrul, 2025-01-07:
Greetings! I have a evcc + TeslaBleHttpProxy setup that I'm having trouble with. Everything is running in a linux VM on proxmox. The VM runs docker and is running both evcc and TeslaBleHttpProxy in separate containers (via docker compose). I also have telegraf installed to monitor the memory/cpu usage of both evcc and TeslaBleHttpProxy. I am using the evcc config that is suggested in the readme.

I've noticed after a while the evcc requests to the proxy start to time out, it looks like this in the logs:
```
[lp-1 ] ERROR 2025/01/07 22:55:43 charger status: Get "http://10.0.0.104:8080/api/1/vehicles/5YJ3E1Exxxxxxxxxx/vehicle_data?endpoints=charge_state": context deadline exceeded (Client.Timeout exceeded while awaiting headers)
```

When this happens, I see the TeslaBleHttpProxy start to consume all the memory in the VM plus the CPU usage jumps significantly (green is memory, blue is CPU):

[image]

Below are the logs from TeslaBleHttpProxy from the time before and during this resource spike:

```
2025/01/07 22:48:40 INFO connecting to Vehicle ...
2025/01/07 22:48:40 DEBU trying connecting to vehicle attempt=1
2025/01/07 22:48:40 DEBU connecting to vehicle (A)...
2025/01/07 22:48:42 DEBU create vehicle object ...
2025/01/07 22:48:42 DEBU connecting to vehicle (B)...
2025/01/07 22:48:42 DEBU start VCSEC session...
2025/01/07 22:48:43 DEBU car successfully wakeup
2025/01/07 22:48:43 DEBU start Infotainment session...
2025/01/07 22:48:44 INFO connection established
2025/01/07 22:48:44 INFO sending command=vehicle_data body=map[endpoints:[charge_state]]
2025/01/07 22:48:44 DEBU get: charge_state
2025/01/07 22:48:46 INFO successfully executed command=vehicle_data body=map[endpoints:[charge_state]]
2025/01/07 22:48:46 INFO sending command=vehicle_data body=map[endpoints:[charge_state]]
2025/01/07 22:48:46 DEBU get: charge_state
2025/01/07 22:48:48 INFO successfully executed command=vehicle_data body=map[endpoints:[charge_state]]
2025/01/07 22:48:48 INFO sending command=vehicle_data body=map[endpoints:[charge_state]]
2025/01/07 22:48:48 DEBU get: charge_state
2025/01/07 22:48:49 INFO successfully executed command=vehicle_data body=map[endpoints:[charge_state]]
2025/01/07 22:48:49 INFO sending command=vehicle_data body=map[endpoints:[charge_state]]
2025/01/07 22:48:49 DEBU get: charge_state
2025/01/07 22:48:51 INFO successfully executed command=vehicle_data body=map[endpoints:[charge_state]]
2025/01/07 22:48:51 INFO sending command=vehicle_data body=map[endpoints:[charge_state]]
2025/01/07 22:48:51 DEBU get: charge_state
2025/01/07 22:48:52 INFO successfully executed command=vehicle_data body=map[endpoints:[charge_state]]
2025/01/07 22:49:10 INFO sending command=vehicle_data body=map[endpoints:[charge_state]]
2025/01/07 22:49:10 DEBU get: charge_state
2025/01/07 22:49:12 INFO successfully executed command=vehicle_data body=map[endpoints:[charge_state]]
2025/01/07 22:49:12 INFO sending command=vehicle_data body=map[endpoints:[charge_state]]
2025/01/07 22:49:12 DEBU get: charge_state
2025/01/07 22:49:13 INFO successfully executed command=vehicle_data body=map[endpoints:[charge_state]]
2025/01/07 22:49:13 INFO sending command=vehicle_data body=map[endpoints:[charge_state]]
2025/01/07 22:49:13 DEBU get: charge_state
2025/01/07 22:49:15 INFO successfully executed command=vehicle_data body=map[endpoints:[charge_state]]
2025/01/07 22:49:15 INFO sending command=vehicle_data body=map[endpoints:[charge_state]]
2025/01/07 22:49:15 DEBU get: charge_state
2025/01/07 22:49:16 INFO successfully executed command=vehicle_data body=map[endpoints:[charge_state]]
2025/01/07 22:49:16 DEBU connection Timeout
2025/01/07 22:49:16 DEBU disconnect vehicle (A)
2025/01/07 22:49:16 DEBU close connection (A)
2025/01/07 22:49:40 INFO connecting to Vehicle ...
2025/01/07 22:49:40 DEBU trying connecting to vehicle attempt=1
2025/01/07 22:49:40 DEBU connecting to vehicle (A)...
2025/01/07 22:49:43 DEBU create vehicle object ...
2025/01/07 22:49:43 DEBU connecting to vehicle (B)...
2025/01/07 22:49:43 DEBU start VCSEC session...
2025/01/07 22:49:45 DEBU car successfully wakeup
2025/01/07 22:49:45 DEBU start Infotainment session...
2025/01/07 22:49:46 INFO connection established
2025/01/07 22:49:46 INFO sending command=vehicle_data body=map[endpoints:[charge_state]]
2025/01/07 22:49:46 DEBU get: charge_state
2025/01/07 22:49:48 INFO successfully executed command=vehicle_data body=map[endpoints:[charge_state]]
2025/01/07 22:49:48 INFO sending command=vehicle_data body=map[endpoints:[charge_state]]
2025/01/07 22:49:48 DEBU get: charge_state
2025/01/07 22:49:49 INFO successfully executed command=vehicle_data body=map[endpoints:[charge_state]]
2025/01/07 22:49:49 INFO sending command=vehicle_data body=map[endpoints:[charge_state]]
2025/01/07 22:49:49 DEBU get: charge_state
[... 30 lines cut ...]
2025/01/07 22:51:11 WARN failed to connect to vehicle (A): failed to find BLE beacon for 5YJ3E1Exxxxxxxxxx (S26d2d43805843cc5C): can't scan: received scan response a4:34:f1:91:f0:2c with no associated Advertising Data packet
2025/01/07 22:51:11 INFO retrying in 6 seconds
2025/01/07 22:51:17 DEBU trying connecting to vehicle attempt=3
2025/01/07 22:51:17 DEBU connecting to vehicle (A)...
2025/01/07 22:51:34 ERRO stop retrying after 3 attempts error="failed to connect to vehicle (A): failed to find BLE beacon for 5YJ3E1Exxxxxxxxxx (S26d2d43805843cc5C): can't scan: received scan response a4:34:f1:91:f0:2c with no associated Advertising Data packet"
2025/01/07 22:51:36 INFO connecting to Vehicle ...
2025/01/07 22:51:36 DEBU trying connecting to vehicle attempt=1
2025/01/07 22:51:36 DEBU connecting to vehicle (A)...
2025/01/07 22:51:51 WARN failed to connect to vehicle (A): failed to find BLE beacon for 5YJ3E1Exxxxxxxxxx (S26d2d43805843cc5C): can't scan: received scan response 6b:e4:d5:ef:9f:37 with no associated Advertising Data packet
2025/01/07 22:51:51 INFO retrying in 3 seconds
2025/01/07 22:51:54 DEBU trying connecting to vehicle attempt=2
2025/01/07 22:51:54 DEBU connecting to vehicle (A)...
2025/01/07 22:52:15 WARN failed to connect to vehicle (A): failed to find BLE beacon for 5YJ3E1Exxxxxxxxxx (S26d2d43805843cc5C): can't scan: received scan response 6b:e4:d5:ef:9f:37 with no associated Advertising Data packet
2025/01/07 22:52:15 INFO retrying in 6 seconds
2025/01/07 22:52:21 DEBU trying connecting to vehicle attempt=3
2025/01/07 22:52:21 DEBU connecting to vehicle (A)...
2025/01/07 22:52:49 ERRO stop retrying after 3 attempts error="failed to connect to vehicle (A): failed to find BLE beacon for 5YJ3E1Exxxxxxxxxx (S26d2d43805843cc5C): can't scan: received scan response 6b:e4:d5:ef:9f:37 with no associated Advertising Data packet"
2025/01/07 22:52:50 INFO connecting to Vehicle ...
2025/01/07 22:52:50 DEBU trying connecting to vehicle attempt=1
2025/01/07 22:52:50 DEBU connecting to vehicle (A)...
2025/01/07 22:53:06 WARN failed to connect to vehicle (A): failed to find BLE beacon for 5YJ3E1Exxxxxxxxxx (S26d2d43805843cc5C): can't scan: received scan response 6b:e4:d5:ef:9f:37 with no associated Advertising Data packet
2025/01/07 22:53:06 INFO retrying in 3 seconds
```

It never seems to recover and I need to restart the container to fix it. Please let me know if there is any other info I can provide to help debug this.

skrul, 2025-01-08:
Here is a little more info -- this morning TeslaBleHttpProxy exhibited this behavior when the car left the house and eventually crashed:

[image]

Relevant logs:
```
2025/01/08 15:16:28 INFO connecting to Vehicle ...
2025/01/08 15:16:28 DEBU trying connecting to vehicle attempt=1
2025/01/08 15:16:28 DEBU connecting to vehicle (A)...
2025/01/08 15:16:29 DEBU create vehicle object ...
2025/01/08 15:16:29 DEBU connecting to vehicle (B)...
2025/01/08 15:16:29 DEBU start VCSEC session...
2025/01/08 15:16:30 DEBU car successfully wakeup
2025/01/08 15:16:30 DEBU start Infotainment session...
2025/01/08 15:16:31 INFO connection established
2025/01/08 15:16:31 INFO sending command=vehicle_data body=map[endpoints:[charge_state]]
2025/01/08 15:16:31 DEBU get: charge_state
2025/01/08 15:16:33 INFO successfully executed command=vehicle_data body=map[endpoints:[charge_state]]
2025/01/08 15:16:33 INFO sending command=vehicle_data body=map[endpoints:[charge_state]]
2025/01/08 15:16:33 DEBU get: charge_state
2025/01/08 15:16:35 INFO successfully executed command=vehicle_data body=map[endpoints:[charge_state]]
2025/01/08 15:16:58 INFO sending command=vehicle_data body=map[endpoints:[charge_state]]
2025/01/08 15:16:58 DEBU get: charge_state
2025/01/08 15:16:58 DEBU disconnect vehicle (A)
2025/01/08 15:16:58 DEBU close connection (A)
2025/01/08 15:16:59 INFO connecting to Vehicle ...
2025/01/08 15:16:59 DEBU trying connecting to vehicle attempt=1
2025/01/08 15:16:59 DEBU connecting to vehicle (A)...
2025/01/08 15:17:14 WARN failed to connect to vehicle (A): context deadline exceeded
2025/01/08 15:17:14 INFO retrying in 3 seconds
2025/01/08 15:17:17 DEBU trying connecting to vehicle attempt=2
2025/01/08 15:17:17 DEBU connecting to vehicle (A)...
2025/01/08 15:17:37 WARN failed to connect to vehicle (A): failed to find BLE beacon for 5YJ3E1EB8JF077233 (S26d2d43805843cc5C): can't scan: received scan response cd:75:61:a5:c7:c9 with no associated Advertising Data packet
2025/01/08 15:17:37 INFO retrying in 6 seconds
2025/01/08 15:17:43 DEBU trying connecting to vehicle attempt=3
2025/01/08 15:17:43 DEBU connecting to vehicle (A)...
2025/01/08 15:18:03 ERRO stop retrying after 3 attempts error="failed to connect to vehicle (A): failed to find BLE beacon for 5YJ3E1EB8JF077233 (S26d2d43805843cc5C): can't scan: received scan response cd:75:61:a5:c7:c9 with no associated Advertising Data packet"
2025/01/08 15:18:04 INFO connecting to Vehicle ...
2025/01/08 15:18:04 DEBU trying connecting to vehicle attempt=1
2025/01/08 15:18:04 DEBU connecting to vehicle (A)...
2025/01/08 15:18:28 WARN failed to connect to vehicle (A): failed to find BLE beacon for 5YJ3E1EB8JF077233 (S26d2d43805843cc5C): can't scan: received scan response cd:75:61:a5:c7:c9 with no associated Advertising Data packet
2025/01/08 15:18:28 INFO retrying in 3 seconds
...
2025/01/08 16:30:13 INFO connecting to Vehicle ...
2025/01/08 16:30:13 DEBU trying connecting to vehicle attempt=1
2025/01/08 16:30:13 DEBU connecting to vehicle (A)...
panic: runtime error: invalid memory address or nil pointer dereference
[signal SIGSEGV: segmentation violation code=0x1 addr=0x0 pc=0x8c4d34]

goroutine 6 [running]:
github.com/go-ble/ble/linux/hci.(*Advertisement).setScanResponse(...)
	/go/pkg/mod/github.com/wimaha/ble_!ble!connect!fix@v0.0.0-20240822192426-3f74826c1268/linux/hci/adv.go:42
github.com/go-ble/ble/linux/hci.(*HCI).handleLEAdvertisingReport(0xc00009c680, {0xc00023d803, 0xc, 0xc})
	/go/pkg/mod/github.com/wimaha/ble_!ble!connect!fix@v0.0.0-20240822192426-3f74826c1268/linux/hci/hci.go:431 +0x3d4
github.com/go-ble/ble/linux/hci.(*HCI).handleLEMeta(0x46a07d?, {0xc00023d803, 0xc, 0xc})
	/go/pkg/mod/github.com/wimaha/ble_!ble!connect!fix@v0.0.0-20240822192426-3f74826c1268/linux/hci/hci.go:398 +0x6d
github.com/go-ble/ble/linux/hci.(*HCI).handleEvt(0xc00009c680, {0xc00023d801, 0xe, 0xe})
	/go/pkg/mod/github.com/wimaha/ble_!ble!connect!fix@v0.0.0-20240822192426-3f74826c1268/linux/hci/hci.go:386 +0x179
github.com/go-ble/ble/linux/hci.(*HCI).handlePkt(0xc0000b12c0?, {0xc00023d800?, 0x1000?, 0xc000208000?})
	/go/pkg/mod/github.com/wimaha/ble_!ble!connect!fix@v0.0.0-20240822192426-3f74826c1268/linux/hci/hci.go:351 +0x17f
github.com/go-ble/ble/linux/hci.(*HCI).sktLoop(0xc00009c680)
	/go/pkg/mod/github.com/wimaha/ble_!ble!connect!fix@v0.0.0-20240822192426-3f74826c1268/linux/hci/hci.go:319 +0xff
created by github.com/go-ble/ble/linux/hci.(*HCI).Init in goroutine 3
	/go/pkg/mod/github.com/wimaha/ble_!ble!connect!fix@v0.0.0-20240822192426-3f74826c1268/linux/hci/hci.go:153 +0x5c5
2025/01/08 16:30:15 INFO TeslaBleHttpProxy 2.0.2 is loading ...
2025/01/08 16:30:15 DEBU LogLevel set to debug
2025/01/08 16:30:15 INFO TeslaBleHttpProxy httpListenAddress=:8080
2025/01/08 16:30:15 DEBU privateKeyFile loaded
2025/01/08 16:30:15 INFO BleControl initialized
2025/01/08 16:30:15 INFO TeslaBleHttpProxy is running!
2025/01/08 16:30:28 INFO connecting to Vehicle ...
2025/01/08 16:30:28 DEBU trying connecting to vehicle attempt=1
2025/01/08 16:30:28 DEBU connecting to vehicle (A)...
2025/01/08 16:30:43 WARN failed to connect to vehicle (A): context deadline exceeded
2025/01/08 16:30:43 INFO retrying in 3 seconds
2025/01/08 16:30:46 DEBU trying connecting to vehicle attempt=2
2025/01/08 16:30:46 DEBU connecting to vehicle (A)...
2025/01/08 16:30:48 can't accept: listner timed out
2025/01/08 16:31:01 WARN failed to connect to vehicle (A): context deadline exceeded
2025/01/08 16:31:01 INFO retrying in 6 seconds
2025/01/08 16:31:07 DEBU trying connecting to vehicle attempt=3
2025/01/08 16:31:07 DEBU connecting to vehicle (A)...
2025/01/08 16:31:22 ERRO stop retrying after 3 attempts error="failed to connect to vehicle (A): context deadline exceeded"
```

wimaha, 2025-01-11:
That looks like a bug in the Go Bluetooth package. Requires further investigation.

skrul, 2025-01-12:
So I enabled some more debugging by calling `ble.SetDebugLog()` (from the `vehicle-command` package) on startup and managed to capture the moment when things spun out of control (full log below). These log lines repeat in a tight loop:

```
Reusing existing BLE device
Searching for BLE beacon S26d2d43805843cc5C...
BLE connection attempt failed: failed to find BLE beacon for 5YJ3E1EB8JF077233 (S26d2d43805843cc5C): can't scan: received scan response 60:10:98:c1:06:98 with no associated Advertising Data packet
```
This infinite `for` loop in the `vehicle-command` package is pretty sus https://github.com/teslamotors/vehicle-command/blob/main/pkg/connector/ble/ble.go#L130

I'll continue to investigate...

```
{"log":"2025/01/12 00:04:19 INFO connecting to Vehicle ...\n","stream":"stderr","time":"2025-01-11T23:04:19.910961044Z"}
{"log":"2025/01/12 00:04:19 DEBU trying connecting to vehicle attempt=1\n","stream":"stderr","time":"2025-01-11T23:04:19.910996239Z"}
{"log":"2025/01/12 00:04:19 DEBU connecting to vehicle (A)...\n","stream":"stderr","time":"2025-01-11T23:04:19.911008109Z"}
{"log":"2025-01-12T00:04:19+01:00 [debug] Reusing existing BLE device\n","stream":"stderr","time":"2025-01-11T23:04:19.911082279Z"}
{"log":"2025-01-12T00:04:19+01:00 [debug] Searching for BLE beacon S26d2d43805843cc5C...\n","stream":"stderr","time":"2025-01-11T23:04:19.911091327Z"}
{"log":"2025-01-12T00:04:20+01:00 [warn ] BLE connection attempt failed: failed to find BLE beacon for 5YJ3E1EB8JF077233 (S26d2d43805843cc5C): can't dial: can't dial: receiv
ed scan response 60:10:98:c1:06:98 with no associated Advertising Data packet\n","stream":"stderr","time":"2025-01-11T23:04:20.050659487Z"}
{"log":"2025-01-12T00:04:20+01:00 [debug] Reusing existing BLE device\n","stream":"stderr","time":"2025-01-11T23:04:20.050689462Z"}
{"log":"2025-01-12T00:04:20+01:00 [debug] Searching for BLE beacon S26d2d43805843cc5C...\n","stream":"stderr","time":"2025-01-11T23:04:20.050695201Z"}
{"log":"2025-01-12T00:04:20+01:00 [warn ] BLE connection attempt failed: failed to find BLE beacon for 5YJ3E1EB8JF077233 (S26d2d43805843cc5C): can't scan: received scan response 60:10:98:c1:06:98 with no associated Advertising Data packet\n","stream":"stderr","time":"2025-01-11T23:04:20.05074225Z"}
{"log":"2025-01-12T00:04:20+01:00 [debug] Reusing existing BLE device\n","stream":"stderr","time":"2025-01-11T23:04:20.050762181Z"}
{"log":"2025-01-12T00:04:20+01:00 [debug] Searching for BLE beacon S26d2d43805843cc5C...\n","stream":"stderr","time":"2025-01-11T23:04:20.050768482Z"}
{"log":"2025-01-12T00:04:20+01:00 [warn ] BLE connection attempt failed: failed to find BLE beacon for 5YJ3E1EB8JF077233 (S26d2d43805843cc5C): can't scan: received scan response 60:10:98:c1:06:98 with no associated Advertising Data packet\n","stream":"stderr","time":"2025-01-11T23:04:20.050828Z"}
{"log":"2025-01-12T00:04:20+01:00 [debug] Reusing existing BLE device\n","stream":"stderr","time":"2025-01-11T23:04:20.050836273Z"}
{"log":"2025-01-12T00:04:20+01:00 [debug] Searching for BLE beacon S26d2d43805843cc5C...\n","stream":"stderr","time":"2025-01-11T23:04:20.050841228Z"}
{"log":"2025-01-12T00:04:20+01:00 [warn ] BLE connection attempt failed: failed to find BLE beacon for 5YJ3E1EB8JF077233 (S26d2d43805843cc5C): can't scan: received scan response 60:10:98:c1:06:98 with no associated Advertising Data packet\n","stream":"stderr","time":"2025-01-11T23:04:20.050845921Z"}
{"log":"2025-01-12T00:04:20+01:00 [debug] Reusing existing BLE device\n","stream":"stderr","time":"2025-01-11T23:04:20.050850743Z"}
{"log":"2025-01-12T00:04:20+01:00 [debug] Searching for BLE beacon S26d2d43805843cc5C...\n","stream":"stderr","time":"2025-01-11T23:04:20.050855109Z"}
{"log":"2025-01-12T00:04:20+01:00 [warn ] BLE connection attempt failed: failed to find BLE beacon for 5YJ3E1EB8JF077233 (S26d2d43805843cc5C): can't scan: received scan response 60:10:98:c1:06:98 with no associated Advertising Data packet\n","stream":"stderr","time":"2025-01-11T23:04:20.05090224Z"}
... same three lines repeat forever
```

skrul, 2025-01-12:
Filed a bug with `vehicle-command` here https://github.com/teslamotors/vehicle-command/issues/346

skrul, 2025-01-26:
@wimaha so I wound up switching to different hardware and this problem went away. Def not a TeslaBleHttpProxy bug so I'll close it.

closed 2025-01-26 by skrul

KNZZZZZZZ, 2025-04-18:
Hi, I have the exact same problem, after a restart of the container it works again.
Are you sure it is a hardware problem?
I am currently using a bluetooth/wifi adapter (Realtek 802.11ax WLAN Adapter).
What hardware solved this?
Thanks!