wimaha/TeslaBleHttpProxy #77: Implement TWC vitals API

skrul, 2025-01-08:
evcc now has direct support for configuring a tesla vehicle using TeslaBleHttpProxy (thanks @wimaha!), but those the use the mobile charger (rather than a TWC) still need to configure a custom charger in evcc that uses the TeslaBleHttpProxy endpoints (see my config [here](https://github.com/skrul/evcc-config/blob/main/evcc.yaml.template#L75)).

I was wondering if TeslaBleHttpProxy has enough information in order to implement the TWC vitals API? If so, mobile charger users could use the [built in evcc TWC charger support](https://github.com/evcc-io/evcc/blob/master/charger/twc3.go), but configure it to talk to TeslaBleHttpProxy instead of a real TWC. Much of the functionality of the built-in TWC support delegates to the vehicle, but it does make use of the following fields from the vitals response:

- contactor_closed
- vehicle_connected
- session_energy_wh
- grid_v
- currentA_a
- currentB_a
- currentC_a
- voltageA_v
- voltageB_v
- voltageC_v

Unfortunately I don't have a TWC so I don't know if these fields map to what is available through TeslaBleHttpProxy. It seems like this should be possible since I am able to create a custom charger in evcc that uses the TeslaBleHttpProxy endpoints.

@wimaha if the mappings exist, I can do the implementation if you think this would be a useful addition to the project!

wimaha, 2025-01-11:
Via ble we don't have the current and voltage for each phase. We only have the sum. This is maybe no issue for the use case.
I'm open for this enhancement!

skrul, 2025-01-11:
@wimaha thanks! I think it could work although I'm not sure if it is the right thing to do. Alternatively I could just contribute my charger config to evcc as a template. One advantage to using the built-in TWC support is that it would make fewer requests to the proxy -- when using the custom charger config, evcc will query the proxy separately for each field, where the TWC plug in would just make one for the various fields. This excessive querying of the proxy could be helped with adding a short cache inside the proxy though.

skrul, 2025-01-11:
Oh I see there is already work being done on a cache here https://github.com/wimaha/TeslaBleHttpProxy/issues/80 fantastic!

MrBLJ, 2025-01-12:
Just my 2 cents since I'm very interested in this topic. I've been using [twc3teslamate](https://github.com/Haveacry/twc3teslamate) so far, to use my mobile charger with EVCC. From my limited understanding, it uses data from teslamate to emulate a TWC.

Having to rely on another service increases the risk of issues, so it'd be best to replicate this functionality using TeslaBleHttpProxy !

skrul, 2025-01-12:
@MrBLJ consider using my custom charger definition for your evcc config https://github.com/skrul/evcc-config/blob/main/evcc.yaml.template#L78 -- this has worked pretty well for me so far. I may just contribute this to evcc rathe than create the vitals API. Please let me know if this is helpful!

skrul, 2025-01-12:
@wimaha @MrBLJ FYI https://github.com/evcc-io/evcc/pull/18194

MrBLJ, 2025-01-14:
> @MrBLJ consider using my custom charger definition for your evcc config https://github.com/skrul/evcc-config/blob/main/evcc.yaml.template#L78 -- this has worked pretty well for me so far. I may just contribute this to evcc rathe than create the vitals API. Please let me know if this is helpful!

Thanks a lot for sharing. I've implemented this on my end. I see that you are currently discussing about turning this into an actual EVCC template, this would be great ! Let me know if I can be of any help, for example if you need logs or data.

MrBLJ, 2025-01-20:
> [@MrBLJ](https://github.com/MrBLJ) consider using my custom charger definition for your evcc config https://github.com/skrul/evcc-config/blob/main/evcc.yaml.template#L78 -- this has worked pretty [...]

I've been using your charger definition for a few days now. One issue : my cars do not go to sleep anymore.

I have a 3 and a Y. I switched the 3 at first using your charger config :
```
- name: my_charger
    type: custom
    power:
      source: http
      uri: "${TESLA_BLE_PROXY}:8080/api/1/vehicles/${TESLA_VIN}/vehicle_data?endpoints=charge_state"
      method: GET
      jq: .response.response.charge_state.charger_voltage * .response.response.charge_state.charger_actual_current
      timeout: 30s
    status:
      source: http
      uri: "${TESLA_BLE_PROXY}:8080/api/1/vehicles/${TESLA_VIN}/vehicle_data?endpoints=charge_state"
      method: GET
      jq: >
        if (.response.response.charge_state.charging_state == "Charging" 
            and .response.response.charge_state.charger_power > 0) then "C"
        elif (.response.response.charge_state.charging_state == "Charging") then "B"
        elif (.response.response.charge_state.charging_state == "Stopped") then "B"
        elif (.response.response.charge_state.charging_state == "NoPower") then "B"
        elif (.response.response.charge_state.charging_state == "Complete") then "B"
        elif (.response.response.charge_state.charging_state == "Starting") then "B"
        else "A" end
      timeout: 30s
    enabled:
      source: http
      uri: "${TESLA_BLE_PROXY}:8080/api/1/vehicles/${TESLA_VIN}/vehicle_data?endpoints=charge_state"
      method: GET
      jq: (.response.response.charge_state.charging_state == "Charging" 
           and .response.response.charge_state.charger_power > 0)
      timeout: 30s
      cache: 25s
    enable:
      source: http
      uri: "${TESLA_BLE_PROXY}:8080/api/1/vehicles/${TESLA_VIN}/command/{{if .enable}}charge_start{{else}}charge_stop{{end}}"
      method: POST
      body: ""
      timeout: 30s
      cache: 25s
    maxcurrent:
      source: http
      uri: "${TESLA_BLE_PROXY}:8080/api/1/vehicles/${TESLA_VIN}/command/set_charging_amps"
      method: POST
      body: '{"charging_amps": "{{.maxcurrent}}"}'
      timeout: 30s
```

(all fields where adapted).

The charger works fine, no issues with it. Seeing that the model 3 would not go to sleep, I reverted back to the original config (using data from Teslamate) which solved the issue. I thought at first it could be vehicle related, so I switched the charger for the Y, and got the same result.

So somehow, on my end, this causes trouble. I have no other 3rd party services that connect to my car (no widget, no app on my watch or whatever).

I can't find a way to determine what is causing my car to stay awake, I'll look into that.

wimaha, 2025-01-20:
Since evcc must poll some status values, the car will be always await with this config.

I think it is possible to change that with the right configuration.

Unfortunately I can look into this from mid February because I’m on a long vacation trip right know.

giss69, 2025-02-10:
@MrBLJ 
I've the same problem, evcc+[TeslaBleHttpProxy] and two cars and they don't sleep anymore.
is hier any chance to fix that because for the time being is it useless.
however I'm using recommended config:
```
vehicles:
  - name: tesla_white
    type: template
    template: tesla-ble
    title: tesla_white
    url: http://192.168.1.33
    port: 8080 # Wird in der Benutzeroberfläche angezeigt # Optional
    vin: XPxxxxxxxxxx # Erforderlich, wenn mehrere Fahrzeuge des Herstellers vorhanden sind # Optional
    capacity: 79 # Akku-Kapazität in kWh # Optiona
  - name: tesla_grey
    type: template
    template: tesla-ble
    title: tesla_grey
    url: http://192.168.1.33
    port: 8080 # Wird in der Benutzeroberfläche angezeigt # Optional
    vin: LRxxxxxxxxxx # Erforderlich, wenn mehrere Fahrzeuge des Herstellers vorhanden sind # Optional
    capacity: 79 # Akku-Kapazität in kWh # Optional
# site describes the EVU connection, PV and home battery
```

giss69, 2025-02-10:
@MrBLJ 
do you defined charger url as TeslaBleHttpProxy? and what port 8001 and 8002 are meaning?

MrBLJ, 2025-02-10:
> [@MrBLJ](https://github.com/MrBLJ) do you defined charger url as TeslaBleHttpProxy? and what port 8001 and 8002 are meaning?

Check the url I specified in my config : `# See also : https://github.com/Haveacry/twc3teslamate`

[This is a pretty nice tool](https://github.com/Haveacry/twc3teslamate). I have two containers running, one for each car, on different ports. This emulates two tesla wall chargers, although I'm using a mobile connector and a dumb wallbox.

skrul, 2025-02-10:
I've been going down this road over on the `evcc` repo and my current sense is that it is not possible to use `TeslaBleHttpProxy` as an evcc "charger" without constantly waking up the car (see https://github.com/evcc-io/evcc/pull/18194#issuecomment-2632811183). I've since purchased a TWC and have a similar setup as @MrBLJ and it works great.

giss69, 2025-02-11:
@MrBLJ 

it works! Thank you very much!
when teslamate won't work anymore because of the new fleet api it stops to work as well?

MrBLJ, 2025-02-13:
> when teslamate won't work anymore because of the new fleet api it stops to work as well?

That's the current concern, but so far it looks like Teslamate could operate with the new fleet API pricing, using the base plan. Guess we'll see.

giss69, 2025-02-15:
Hi,

today we have sun and I tried to charge with my new setting without success. 
my twc3 wasn't recognised as connected. 
`{"contactor_closed":false,"vehicle_connected":false,"session_s":0,"grid_v":0.0,"grid_hz":49.828,"vehicle_current_a":0.0,"currentA_a":0.0,"currentB_a":0.0,"currentC_a":0.0,"currentN_a":0.0,"voltageA_v":0.0,"voltageB_v":0.0,"voltageC_v":0.0,"relay_coil_v":11.9,"pcba_temp_c":7.4,"handle_temp_c":1.8,"mcu_temp_c":13.0,"uptime_s":26103,"input_thermopile_uv":-176,"prox_v":0.0,"pilot_high_v":11.9,"pilot_low_v":11.8,"session_energy_wh":52260.0,"config_status":5,"evse_state":1,"current_alerts":[]}`

giss69, 2025-02-16:
@MrBLJ 
Hi,

today we have sun and I tried to charge with my new setting without success. 
my twc3 wasn't recognised as connected. 
`{"contactor_closed":false,"vehicle_connected":false,"session_s":0,"grid_v":0.0,"grid_hz":49.828,"vehicle_current_a":0.0,"currentA_a":0.0,"currentB_a":0.0,"currentC_a":0.0,"currentN_a":0.0,"voltageA_v":0.0,"voltageB_v":0.0,"voltageC_v":0.0,"relay_coil_v":11.9,"pcba_temp_c":7.4,"handle_temp_c":1.8,"mcu_temp_c":13.0,"uptime_s":26103,"input_thermopile_uv":-176,"prox_v":0.0,"pilot_high_v":11.9,"pilot_low_v":11.8,"session_energy_wh":52260.0,"config_status":5,"evse_state":1,"current_alerts":[]}`
but if I used direct ip of my twc3 I had all information.
Could you pleas share you docker-compose to setup teslamate with twc3teslamate?

Thank you!

MrBLJ, 2025-02-24:
> Since evcc must poll some status values, the car will be always await with this config.
> 
> I think it is possible to change that with the right configuration.

@wimaha, looks like [this has been implemented here](https://github.com/tesla-local-control/tesla_ble_mqtt_docker?tab=readme-ov-file#note-regarding-polling--reading-of-state).

MrBLJ, 2025-02-24:
@giss69 :

```yaml
services:
  twc3teslamate_model3:
    container_name: twc3teslamate_model3
    image: ghcr.io/haveacry/twc3teslamate
    environment:
      - MQTT_HOST=192.168.1.51
      - MQTT_USERNAME=TeslaMate
      - MQTT_PASSWORD=X
      - TESLAMATE_CAR_ID=2
    restart: unless-stopped
    ports:
      - 8001:80
  twc3teslamate_modelY:
    container_name: twc3teslamate_modelY
    image: ghcr.io/haveacry/twc3teslamate
    environment:
      - MQTT_HOST=192.168.1.51
      - MQTT_USERNAME=TeslaMate
      - MQTT_PASSWORD=X
      - TESLAMATE_CAR_ID=1
    restart: unless-stopped
    ports:
      - 8002:80
```