wimaha/TeslaBleHttpProxy #80: Cache control

Lenart12, 2025-01-09:
Previous discussion here: https://github.com/evcc-io/evcc/issues/18142

Add cache control headers to vehicle_data endpoints. Probably as simple as adding header `Cache-Control: public, max-age=${cache_max_age}` and associated configuration for max age.

andig, 2025-01-10:
Looking at the template:

    uri: {{ .url }}:{{ .port }}/api/1/vehicles/{{ .vin }}/vehicle_data?endpoints=charge_state
    method: GET

It might be nice to add an ETag for the contents instead of maxage. Ymmv.

Lenart12, 2025-01-10:
I dont get yes how ETag would help. I'm hoping maybe you can explain what you mean because with my understanding Etag would be used as either:
* (ETag generated from time/random) the proxy would need to take a look when the tag was created and compare it to some set internal timeout (effectivly making it the same as max-age)
* (ETag generated from response hash) the proxy would need to send another BLE request to the car, generate a new etag computed from the answer and compare it to the requested one (making caching pointless since BLE is the slowest part, not http transport)

wimaha, 2025-01-11:
Thanks for the issue.
In my opinion, the ETag does not solve the problem it is intended to address: namely, reducing BLE requests.

So we may use only max-age.

wimaha, 2025-01-12:
Implemented with #81

closed 2025-01-12 by wimaha