wimaha/TeslaBleHttpProxy #82: Home Assistant Add-On

stumbaumr, 2025-01-10:
Would you be willing to add configuration files to make this docker container available as a Home Assistant add-on?
https://developers.home-assistant.io/docs/add-ons/

Some people run evcc.io as a Home Assistant Add-on - it would be conveniant to just be able to run TeslaBleHttpProxy as an Add-On as well.

Trunks1982, 2025-01-21:
Das würde mir auch gefallen!!👌

Lenart12, 2025-01-29:
I created a similar repo [TeslaBle2Mqtt-addon](https://github.com/Lenart12/TeslaBle2Mqtt-addon) that starts this proxy together with [TeslaBle2Mqtt](https://github.com/Lenart12/TeslaBle2Mqtt). Someone could make an addon fairly easily just for this proxy by removing the "2Mqtt" part and its related configuration.

Trunks1982, 2025-01-29:
> Ich habe ein ähnliches Repo [TeslaBle2Mqtt-Addon ](https://github.com/Lenart12/TeslaBle2Mqtt-addon)erstellt, das diesen Proxy zusammen mit [TeslaBle2Mqtt ](https://github.com/Lenart12/TeslaBle2Mqtt)startet. Jemand könnte ziemlich einfach ein Addon nur für diesen Proxy erstellen, indem er den "2Mqtt"-Teil und die zugehörige Konfiguration entfernt.

Hi ich hab gerade das addon installiert.Wie bekomme ich das addon mit tesla ble verbunden?

Trunks1982, 2025-01-29:
Ich habe es gerade mal installiert aber bekomme mit meinem vorhandenen usb blenden keine Verbindung zum Auto.BLE ist aber in ha drin.

Lenart12, 2025-01-29:
Does TeslaBleHttpProxy work on your hardware without the addon, running it directly? This might be problem because you are using usb dongle.

Trunks1982, 2025-01-29:
Also ich nutze im proxmox eigens eine vm nur für reine http ble und da funktioniert es mit einem usb dongle.Ich habe zwei stk.Einmal für ha vm selber und ble http vm auch.Wollte das halt nur in Ha nutzen.

Trunks1982, 2025-01-29:
Ist nicht schlimm das du das nicht weisst.Sicher ist das der usb komplett in der VM durchgereicht wird. Im Linux Vm läuft ja der proxy auch.Frage kann man nicht angeben in der programmierung auf was dr proxy alles zugreifen soll?
Also zb nicht nur hci?

Trunks1982, 2025-01-29:
Wie meinste lokal zu installieren?
In HA OS wüsste ich nicht das ich nen compose  installieren kann.

Trunks1982, 2025-01-29:
sehe nur das im log:    2025/01/29 18:50:24 INFO received command=connection_status
2025/01/29 18:50:44 INFO ECDSA private key generated and saved
2025/01/29 18:50:44 INFO ECDSA public key generated and saved
2025/01/29 18:50:44 INFO BleControl initialized
s6-rc: info: service legacy-services: stopping
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
[18:51:09] INFO: Starting TeslaBle2Mqtt
2025/01/29 18:51:10 INFO Starting TeslaBle2Mqtt
2025/01/29 18:51:10 INFO TeslaBleHttpProxy *undefined* is loading ...
2025/01/29 18:51:10 INFO BleControl initialized

Trunks1982, 2025-01-29:
> Can you build my fork https://github.com/Lenart12/TeslaBleHttpProxy/ in your Linux VM and check if that works? Maybe one of my new commits breaks the proxy on your vm.

Ich hab jetzt ihr repo neu gebildet in meiner normalen vm.Da funktioniert auch das senden des keys zum Auto.

Trunks1982, 2025-01-29:
Jetzt ist die frage ob der dongle schon in ha verwendet werden darf?
Er wird nämlich in der geräte liste angezeigt und verwendet.

Lenart12, 2025-01-29:
Hmm güt news and bad news :D I think we can be fairly certain the problem is with docker. I will try giving full privileges to the test addon with `full_access: true`. Uninstall, check for updates and reinstall please.

[image]

Trunks1982, 2025-01-29:
2025/01/29 19:30:19 INFO Starting TeslaBle2Mqtt
2025/01/29 19:30:19 INFO TeslaBleHttpProxy *undefined* is loading ...
2025/01/29 19:30:19 INFO BleControl initialized
2025/01/29 19:30:19 INFO TeslaBleHttpProxy is running!
2025/01/29 19:30:19 INFO Connected to MQTT client_id=tb2m_XPxxxxxxxxxxxxx
2025/01/29 19:30:19 INFO Connected to MQTT client_id=tb2m
2025/01/29 19:30:19 INFO Publishing topic=tb2m/tb2m/uptime/state access path=uptime state=0 old_state=""
2025/01/29 19:30:19 INFO Publishing topic=tb2m/status access path=status state=online old_state=""
2025/01/29 19:30:19 INFO received command=connection_status
2025/01/29 19:30:20 INFO received command=body_controller_state
2025/01/29 19:30:21 INFO connecting to Vehicle ...
2025/01/29 19:30:32 WARN failed to perform handshake with vehicle (A): vehicle rejected request: your public key has not been paired with the vehicle
2025/01/29 19:30:32 INFO retrying in 3 seconds
2025/01/29 19:30:36 ERRO can't connect to vehicle error="vehicle not in range: ble: failed to scan for XPxxxxxxxxxxxx: context deadline exceeded"
2025/01/29 19:30:40 can't accept: listner timed out
2025/01/29 19:30:43 INFO Key-Request connection established
2025/01/29 19:30:43 WARN failed to wake up car: cannot send authenticated command before establishing a vehicle session
2025/01/29 19:30:43 INFO retrying in 3 seconds
2025/01/29 19:30:46 WARN failed to wake up car: cannot send authenticated command before establishing a vehicle session
2025/01/29 19:30:46 INFO retrying in 6 seconds
2025/01/29 19:30:52 WARN max retries reached command=add-key-request body=map[] err="failed to wake up car: cannot send authenticated command before establishing a vehicle session"
 L

[image]

Fehler bleibt bestehen.

Trunks1982, 2025-01-29:
[image]

Das wird mir aus den Diagonosedaten von HA Bluetooth angezeigt wenn ich die json runterlade.

Trunks1982, 2025-01-29:
025/01/29 19:42:13 DEBU disconnect vehicle (B)
2025/01/29 19:42:13 DEBU close connection (B)
2025/01/29 19:42:13 WARN failed to perform handshake with vehicle (A): vehicle rejected request: your public key has not been paired with the vehicle
2025/01/29 19:42:13 INFO retrying in 3 seconds
2025/01/29 19:42:16 DEBU trying connecting to vehicle attempt=2
2025/01/29 19:42:16 DEBU connecting to vehicle (A)...
2025/01/29 19:42:16 DEBU scan for vehicle ...
2025/01/29 19:42:16 DEBU beacon found localName=S1f367be5086602d3C addr=74:b8:39:52:93:6c rssi=-76
2025/01/29 19:42:16 DEBU dialing to vehicle ...
2025/01/29 19:42:17 DEBU create vehicle object ...
2025/01/29 19:42:17 DEBU connecting to vehicle (B)...
2025/01/29 19:42:17 DEBU start VCSEC session...
2025/01/29 19:42:17 DEBU disconnect vehicle (B)
2025/01/29 19:42:17 DEBU close connection (B)
2025/01/29 19:42:17 WARN failed to perform handshake with vehicle (A): vehicle rejected request: your public key has not been paired with the vehicle
2025/01/29 19:42:17 INFO retrying in 6 seconds
2025/01/29 19:42:23 DEBU trying connecting to vehicle attempt=3
2025/01/29 19:42:23 DEBU connecting to vehicle (A)...
2025/01/29 19:42:23 DEBU scan for vehicle ...
2025/01/29 19:42:24 ERRO can't connect to vehicle error="vehicle not in range: ble: failed to scan for XP7----------: context deadline exceeded"
2025/01/29 19:42:24 DEBU connecting to Vehicle done
2025/01/29 19:42:24 DEBU response command=body_controller_state status=503 result=false reason="vehicle not in range: ble: failed to scan for XP---------------: context deadline exceeded"
2025/01/29 19:42:24 DEBU Failed to get state error="failed to get body controller state: command failed: vehicle not in range: ble: failed to scan for XP--------------: context deadline exceeded"
2025/01/29 19:42:24 DEBU Waiting for next poll interval interval=90
2025/01/29 19:42:25 DEBU waiting for command
2025/01/29 19:42:28 can't accept: listner timed out
2025/01/29 19:43:08 DEBU connecting to vehicle (A)...
2025/01/29 19:43:08 DEBU scan for vehicle ...
2025/01/29 19:43:08 DEBU beacon found localName=S1f367be5086602d3C addr=74:b8:39:52:93:6c rssi=-76
2025/01/29 19:43:08 DEBU dialing to vehicle ...
2025/01/29 19:43:13 DEBU create vehicle object ...
2025/01/29 19:43:13 DEBU connecting to vehicle (B)...
2025/01/29 19:43:13 INFO Key-Request connection established
2025/01/29 19:43:13 DEBU sending command=add-key-request body=map[]
2025/01/29 19:43:13 WARN failed to wake up car: cannot send authenticated command before establishing a vehicle session
2025/01/29 19:43:13 INFO retrying in 3 seconds
2025/01/29 19:43:16 WARN failed to wake up car: cannot send authenticated command before establishing a vehicle session
2025/01/29 19:43:16 INFO retrying in 6 seconds
2025/01/29 19:43:22 WARN max retries reached command=add-key-request body=map[] err="failed to wake up car: cannot send authenticated command before establishing a vehicle session"
2025/01/29 19:43:22 DEBU disconnect vehicle (A)
2025/01/29 19:43:23 DEBU close connection (A)

Trunks1982, 2025-01-29:
2025/01/29 19:47:34 DEBU received handler=ProxyCommand method=GET url=/api/proxy/1/vehicles/X-----------/body_controller_state from=127.0.0.1:35464
2025/01/29 19:47:34 INFO received command=body_controller_state
2025/01/29 19:47:35 DEBU waiting for command
2025/01/29 19:47:35 INFO connecting to Vehicle ...
2025/01/29 19:47:35 DEBU trying connecting to vehicle attempt=1
2025/01/29 19:47:35 DEBU connecting to vehicle (A)...
2025/01/29 19:47:35 DEBU scan for vehicle ...
2025/01/29 19:47:36 DEBU beacon found localName=S1f367be5086602d3C addr=74:b8:39:52:93:6c rssi=-86
2025/01/29 19:47:36 DEBU dialing to vehicle ...
2025/01/29 19:47:36 DEBU create vehicle object ...
2025/01/29 19:47:36 DEBU connecting to vehicle (B)...
2025/01/29 19:47:36 DEBU start VCSEC session...
2025/01/29 19:47:37 DEBU disconnect vehicle (B)
2025/01/29 19:47:37 DEBU close connection (B)
2025/01/29 19:47:37 WARN failed to perform handshake with vehicle (A): vehicle rejected request: your public key has not been paired with the vehicle
2025/01/29 19:47:37 INFO retrying in 3 seconds
2025/01/29 19:47:40 DEBU trying connecting to vehicle attempt=2
2025/01/29 19:47:40 DEBU connecting to vehicle (A)...
2025/01/29 19:47:40 DEBU scan for vehicle ...
2025/01/29 19:47:40 DEBU beacon found localName=S1f367be5086602d3C addr=74:b8:39:52:93:6c rssi=-86
2025/01/29 19:47:40 DEBU dialing to vehicle ...
2025/01/29 19:47:41 DEBU create vehicle object ...
2025/01/29 19:47:41 DEBU connecting to vehicle (B)...
2025/01/29 19:47:41 DEBU start VCSEC session...
2025/01/29 19:47:41 DEBU disconnect vehicle (B)
2025/01/29 19:47:41 DEBU close connection (B)
2025/01/29 19:47:41 WARN failed to perform handshake with vehicle (A): vehicle rejected request: your public key has not been paired with the vehicle
2025/01/29 19:47:41 INFO retrying in 6 seconds
2025/01/29 19:47:47 DEBU trying connecting to vehicle attempt=3
2025/01/29 19:47:47 DEBU connecting to vehicle (A)...
2025/01/29 19:47:47 DEBU scan for vehicle ...
2025/01/29 19:47:47 DEBU beacon found localName=S1f367be5086602d3C addr=74:b8:39:52:93:6c rssi=-86
2025/01/29 19:47:47 DEBU dialing to vehicle ...
2025/01/29 19:47:50 DEBU create vehicle object ...
2025/01/29 19:47:50 DEBU connecting to vehicle (B)...
2025/01/29 19:47:50 DEBU start VCSEC session...

Ich sitze im Auto und schick es los.

Lenart12, 2025-01-29:
```
beacon found localName=S1f367be5086602d3C addr=74:b8:39:52:93:6c rssi=-76
```
This is good. It means your usb dongle is working, so we solved the container issue and run in to a new one. I think I broke adding keys in my commits... I will look at the code and see where it goes wrong.

Trunks1982, 2025-01-29:
Super das es mit dem dongle jetzt schonmal funktionieren würde.Bin für weitere tests bereit.Musst nur bescheid sagen.

Trunks1982, 2025-01-29:
Super Schlüssel erstellen hat geklappt. Jetzt bekomme ich aber noch keine Daten in Mqtt.

Trunks1982, 2025-01-29:
2025/01/29 20:18:20 DEBU waiting for command
2025/01/29 20:18:28 DEBU received handler=VehicleEndpoint method=GET url="/api/1/vehicles/XP7YGCES9RB368880/vehicle_data?endpoints=charge_state" from=192.168.0.124:48442
2025/01/29 20:18:28 INFO received command=vehicle_data body=map[endpoints:[charge_state]]
2025/01/29 20:18:28 INFO connecting to Vehicle ...
2025/01/29 20:18:28 DEBU trying connecting to vehicle attempt=1
2025/01/29 20:18:28 DEBU connecting to vehicle (A)...
2025/01/29 20:18:28 DEBU scan for vehicle ...
2025/01/29 20:18:28 DEBU beacon found localName=S1f367be5086602d3C addr=74:b8:39:52:93:6c rssi=-76
2025/01/29 20:18:28 DEBU dialing to vehicle ...
2025/01/29 20:18:29 DEBU create vehicle object ...
2025/01/29 20:18:29 DEBU connecting to vehicle (B)...
2025/01/29 20:18:29 DEBU start VCSEC session...
2025/01/29 20:18:30 DEBU disconnect vehicle (B)
2025/01/29 20:18:30 DEBU close connection (B)
2025/01/29 20:18:30 WARN failed to perform handshake with vehicle (A): vehicle rejected request: your public key has not been paired with the vehicle
2025/01/29 20:18:30 INFO retrying in 3 seconds
2025/01/29 20:18:33 DEBU trying connecting to vehicle attempt=2
2025/01/29 20:18:33 DEBU connecting to vehicle (A)...
2025/01/29 20:18:33 DEBU scan for vehicle ...
2025/01/29 20:18:34 ERRO can't connect to vehicle error="vehicle not in range: ble: failed to scan for XPxxx: context deadline exceeded"
2025/01/29 20:18:34 DEBU connecting to Vehicle done
2025/01/29 20:18:34 DEBU response command=vehicle_data status=503 result=false reason="vehicle not in range: ble: failed to scan for XP7YGCES9RB368880: context deadline exceeded"
2025/01/29 20:18:34 DEBU received handler=VehicleEndpoint method=GET url="/api/1/vehicles/LRW3E7FS8PC709491/vehicle_data?endpoints=charge_state" from=192.168.0.124:51174
2025/01/29 20:18:34 INFO received command=vehicle_data body=map[endpoints:[charge_state]]
2025/01/29 20:18:35 DEBU waiting for command
2025/01/29 20:18:35 INFO connecting to Vehicle ...
2025/01/29 20:18:35 DEBU trying connecting to vehicle attempt=1
2025/01/29 20:18:35 DEBU connecting to vehicle (A)...
2025/01/29 20:18:35 DEBU scan for vehicle ...
2025/01/29 20:18:36 ERRO can't connect to vehicle error="vehicle not in range: ble: failed to scan for LRWxxxxxxx: context deadline exceeded"
2025/01/29 20:18:36 DEBU connecting to Vehicle done
2025/01/29 20:18:36 DEBU response command=vehicle_data status=503 result=false reason="vehicle not in range: ble: failed to scan for LRW3E7FS8PC709491: context deadline exceeded"
2025/01/29 20:18:37 DEBU waiting for command
2025/01/29 20:18:42 DEBU Publish state loop handler=handler for=tb2m
2025/01/29 20:18:42 INFO Publishing topic=tb2m/tb2m/uptime/state access path=uptime state=90 old_state=0
2025/01/29 20:18:42 DEBU Waiting for next poll interval interval=90
2025/01/29 20:18:44 DEBU received handler=VehicleEndpoint method=GET url="/api/1/vehicles/XP7YGCES9RB368880/vehicle_data?endpoints=charge_state" from=192.168.0.124:48442
2025/01/29 20:18:44 INFO received command=vehicle_data body=map[endpoints:[charge_state]]
2025/01/29 20:18:44 INFO connecting to Vehicle ...
2025/01/29 20:18:44 DEBU trying connecting to vehicle attempt=1
2025/01/29 20:18:44 DEBU connecting to vehicle (A)...
2025/01/29 20:18:44 DEBU scan for vehicle ...
2025/01/29 20:18:44 DEBU beacon found localName=S1f367be5086602d3C addr=74:b8:39:52:93:6c rssi=-84
2025/01/29 20:18:44 DEBU dialing to vehicle ...
2025/01/29 20:18:45 DEBU create vehicle object ...
2025/01/29 20:18:45 DEBU connecting to vehicle (B)...
2025/01/29 20:18:45 DEBU start VCSEC session...

Trunks1982, 2025-01-29:
2025/01/29 20:19:19 DEBU waiting for command
2025/01/29 20:19:19 DEBU response command=connection_status status=200 result=true reason="The command was successfully processed."
2025/01/29 20:19:19 DEBU Getting proxy response url=http://0.0.0.0:5667/api/proxy/1/vehicles/XP7YGCES9RB368880/body_controller_state
2025/01/29 20:19:19 DEBU received handler=ProxyCommand method=GET url=/api/proxy/1/vehicles/XP7YGCES9RB368880/body_controller_state from=127.0.0.1:43058
2025/01/29 20:19:19 INFO received command=body_controller_state
2025/01/29 20:19:20 DEBU waiting for command
2025/01/29 20:19:20 INFO connecting to Vehicle ...
2025/01/29 20:19:20 DEBU trying connecting to vehicle attempt=1
2025/01/29 20:19:20 DEBU connecting to vehicle (A)...
2025/01/29 20:19:20 DEBU scan for vehicle ...
2025/01/29 20:19:21 DEBU beacon found localName=S1f367be5086602d3C addr=74:b8:39:52:93:6c rssi=-84
2025/01/29 20:19:21 DEBU dialing to vehicle ...
2025/01/29 20:19:23 DEBU create vehicle object ...
2025/01/29 20:19:23 DEBU connecting to vehicle (B)...
2025/01/29 20:19:23 DEBU start VCSEC session...
2025/01/29 20:19:23 DEBU disconnect vehicle (B)
2025/01/29 20:19:23 DEBU close connection (B)
2025/01/29 20:19:23 WARN failed to perform handshake with vehicle (A): vehicle rejected request: your public key has not been paired with the vehicle
2025/01/29 20:19:23 INFO retrying in 3 seconds
2025/01/29 20:19:26 DEBU trying connecting to vehicle attempt=2
2025/01/29 20:19:26 DEBU connecting to vehicle (A)...
2025/01/29 20:19:26 DEBU scan for vehicle ...
2025/01/29 20:19:26 DEBU beacon found localName=S1f367be5086602d3C addr=74:b8:39:52:93:6c rssi=-86
2025/01/29 20:19:26 DEBU dialing to vehicle ...
2025/01/29 20:19:27 DEBU create vehicle object ...
2025/01/29 20:19:27 DEBU connecting to vehicle (B)...
2025/01/29 20:19:27 DEBU start VCSEC session...
2025/01/29 20:19:27 DEBU disconnect vehicle (B)
2025/01/29 20:19:27 DEBU close connection (B)
2025/01/29 20:19:27 WARN failed to perform handshake with vehicle (A): vehicle rejected request: your public key has not been paired with the vehicle
2025/01/29 20:19:27 INFO retrying in 6 seconds
2025/01/29 20:19:28 DEBU received handler=VehicleEndpoint method=GET url="/api/1/vehicles/XP7YGCES9RB368880/vehicle_data?endpoints=charge_state" from=192.168.0.124:48442
2025/01/29 20:19:28 INFO received command=vehicle_data body=map[endpoints:[charge_state]]
2025/01/29 20:19:33 DEBU trying connecting to vehicle attempt=3
2025/01/29 20:19:33 DEBU connecting to vehicle (A)...
2025/01/29 20:19:33 DEBU scan for vehicle ...
2025/01/29 20:19:34 DEBU beacon found localName=S1f367be5086602d3C addr=74:b8:39:52:93:6c rssi=-76

Lenart12, 2025-01-29:
Its saying your key is not paired with the vehicle. Maybe delete the keys and retry the pairing process? I'm unsure what would be other problem.

Trunks1982, 2025-01-29:
Er meint glaub mein zweites auto.die vin kommt auch noch vor im log.

Trunks1982, 2025-01-29:
Perfekt jetzt gehts.👍

Trunks1982, 2025-01-29:
[image]

Aber kleine json fehler sind da noch.

Trunks1982, 2025-01-29:
Ist das richtig wenn das Auto in den Ruhe zustand geht das alle Werte in mqtt dann gelöscht werden?

Lenart12, 2025-01-29:
Can you send me more logs of the json errors, its the first time I see them.

Yes, its correct. It has 3 types: no car, car asleep, and car awake. And it will update according to that. Also the app wont keep the car awake. If you want data to repear you need to press the wakeup car button or do it with an automation.

I will remove the testing addon shortly and replace it with a maintained [Full access] version of the addon, you can switch to it then.

Lenart12, 2025-01-30:
Looks like this related issue: https://github.com/noble/noble/issues/864 
Same problem with starting the scan... Is your bluetooth dongle BLE 4.x or 5.x?
Maybe this is also related https://github.com/tesla-local-control/tesla_ble_mqtt_core/issues/125 because this issue is also related to scanning. I guess it could be possible for proxy to detect "Command Disallowed" and restart hci device before retrying, but this would also need a PR in teslamotors/vehicle-command repo to allow restarting device.

wimaha, 2025-01-30:
The Go BLE stack isn't really that good. But what could also help is the new implementation, where connection attempts aren't constantly made when the car isn't there. 

Question: Do you still try to establish a connection even when the car is away?

Lenart12, 2025-01-30:
No, only scanning for beacons for 1 second every 90 seconds by default

Trunks1982, 2025-01-30:
> Looks like this related issue: [noble/noble#864](https://github.com/noble/noble/issues/864) Same problem with starting the scan... Is your bluetooth dongle BLE 4.x or 5.x? Maybe this is also related [tesla-local-control/tesla_ble_mqtt_core#125](https://github.com/tesla-local-control/tesla_ble_mqtt_core/issues/125) because this issue is also related to scanning. I guess it could be possible for proxy to detect "Command Disallowed" and restart hci device before retrying, but this would also need a PR in teslamotors/vehicle-command repo to allow restarting device.
> 
> https://bugzilla.kernel.org/show_bug.cgi?id=202515#c4

Ble 5.x benutze ich.

Trunks1982, 2025-01-30:
Da ist wieder der json Fehler

2025/01/30 13:34:53 INFO successfully executed command=vehicle_data body=map[endpoints:[charge_state]]
2025/01/30 13:34:53 ERRO decoding body err="json: cannot unmarshal bool into Go value of type map[string]interface {}"
2025/01/30 13:35:01 can't accept: listner timed out
2025/01/30 13:35:03 INFO received command=vehicle_data body=map[endpoints:[charge_state]]
2025/01/30 13:35:05 INFO Handling command key=PRESS action=clear_error body=""
2025/01/30 13:35:05 INFO received command=connection_status
2025/01/30 13:35:05 INFO successfully executed command=vehicle_data body=map[endpoints:[charge_state]]
2025/01/30 13:35:05 INFO received command=body_controller_state
2025/01/30 13:35:05 ERRO decoding body err="json: cannot unmarshal bool into Go value of type map[string]interface {}"
2025/01/30 13:35:05 INFO successfully executed command=body_controller_state body=map[]
2025/01/30 13:35:05 INFO received command=vehicle_data body="map[endpoints:[charge_state climate_state]]"
2025/01/30 13:35:08 INFO successfully executed command=vehicle_data body="map[endpoints:[charge_state climate_state]]"
2025/01/30 13:35:08 INFO Publishing topic=tb2m/XP------------------/inside_temperature/state access path=vehicle_data.climate_state.inside_temp state=10.800001 old_state=10.300001
2025/01/30 13:35:18 INFO received command=vehicle_data body=map[endpoints:[charge_state]]
2025/01/30 13:35:18 INFO connecting to Vehicle ...
2025/01/30 13:35:24 INFO connection established
2025/01/30 13:35:26 INFO successfully executed command=vehicle_data body=map[endpoints:[charge_state]]
2025/01/30 13:35:26 ERRO decoding body err="json: cannot unmarshal bool into Go value of type map[string]interface {}"
2025/01/30 13:35:33 INFO received command=vehicle_data body=map[endpoints:[charge_state]]
2025/01/30 13:35:34 INFO successfully executed command=vehicle_data body=map[endpoints:[charge_state]]
2025/01/30 13:36:10 INFO Publishing topic=tb2m/tb2m/uptime/state access path=uptime state=90 old_state=0
2025/01/30 13:36:12 INFO received command=connection_status

Trunks1982, 2025-01-30:
2025/01/30 13:40:40 ERRO decoding body err="json: cannot unmarshal bool into Go value of type map[string]interface {}"
2025/01/30 13:40:48 ERRO decoding body err="json: cannot unmarshal bool into Go value of type map[string]interface {}"

Lenart12, 2025-01-30:
Are you sending any commands to the vehicle, pressing buttons or automation, or just monitoring state?

Trunks1982, 2025-01-30:
2025/01/30 13:48:12 ERRO decoding body err="json: cannot unmarshal bool into Go value of type map[string]interface {}"
2025/01/30 13:48:13 ERRO decoding body err="json: cannot unmarshal bool into Go value of type map[string]interface {}"

Das kam wo ich bei evcc von pv auf min+pv gewechselt bin.

Lenart12, 2025-01-30:
Evcc is sending true to /charge_start command. They need to fix this. Charge start expects no body. https://developer.tesla.com/docs/fleet-api/endpoints/vehicle-commands#charge-start

Trunks1982, 2025-01-30:
Ok dann müsste es schon bei wimaha falsch sein. Ich glaub er ist da nah dran an evcc.

Trunks1982, 2025-01-30:
Ich hab mal frage um zb den bluetooth  und Reichweite noch zu erhöhen, kann man das eigentlich auch mit Bluetooth proxy laufen lassen?

Lenart12, 2025-01-30:
As far as I know probably not. Try setting scan timeout to something larger, 5 seconds maybe. Also check this out but i dont know if it works https://github.com/BogdanDIA/esp32-hci-proxy-addon

Trunks1982, 2025-01-30:
Danke werde ich testen.

Gerade wieder unterwegs gesewesen und er verbindet sich wieder nicht mit dem Auto.

2025/01/30 15:05:13 DEBU received handler=VehicleEndpoint method=GET url="/api/1/vehicles/LRW3E7FS8PC709491/vehicle_data?endpoints=charge_state" from=192.168.0.124:41968
2025/01/30 15:05:13 INFO received command=vehicle_data body=map[endpoints:[charge_state]]
2025/01/30 15:05:13 ERRO Failed to publish state error="failed to get connection status: command failed: ble: failed to scan for XP7YGCES9RB368880: Command Disallowed"
2025/01/30 15:05:13 DEBU Publish state loop handler=per_vehicle for=tb2m_XP7xxxxxxxxx
2025/01/30 15:05:13 DEBU Getting proxy response url=http://0.0.0.0:5667/api/proxy/1/vehicles/XPxxxxxxxxxxxxx/connection_status
2025/01/30 15:05:13 DEBU received handler=ProxyCommand method=GET url=/api/proxy/1/vehicles/XP7YGCES9RB368880/connection_status from=127.0.0.1:43496
2025/01/30 15:05:13 INFO received command=connection_status
2025/01/30 15:05:14 DEBU waiting for command
2025/01/30 15:05:14 INFO connecting to Vehicle ...
2025/01/30 15:05:14 DEBU trying connecting to vehicle attempt=1
2025/01/30 15:05:14 DEBU connecting to vehicle (A)...
2025/01/30 15:05:14 DEBU scan for vehicle ...

Lenart12, 2025-01-30:
@Trunks1982 please update to v0.0.11 as it is now configured with cacheMaxAge and will work better with evcc. Or alternatively install standalone proxy addon from the same repository if mqtt is not important to you. Key pairing should be easier now that I integrated the proxy dashboard as a homeassistant ingress panel (web ui).

Trunks1982, 2025-01-30:
Super danke ich werde es testen.Was macht der cache genau?
Lass den Wert erstmal auf 5s.

Lenart12, 2025-01-30:
When evcc gets data for each sensor, the proxy needs to ask the car for all sensors at once and returns them all at the same time but evcc only keeps one. This makes it so that it caches the response with all sensors for each sensor instead of making the proxy do a car request every time.

Trunks1982, 2025-01-31:
Habs installiert. Aber stoppen kann ich den ladevorgang noch nicht.Was ich nicht verstehe mit dem von wihama hat das funktioniert.Ich konnte ladung stoppen und starten.

Trunks1982, 2025-01-31:
Könnte ja gerne mal nen debug von der eigenen VM für bleiben machen.Dann kann man sehn was er bei wimaha schickt.

Trunks1982, 2025-01-31:
das kommt aus dem originalen.Also mit evcc kann ich umschalten.    

2025/01/31 12:50:40 DEBU car successfully wakeup
2025/01/31 12:50:40 DEBU start Infotainment session...
2025/01/31 12:50:41 INFO connection established
2025/01/31 12:50:41 INFO sending command=vehicle_data body=map[endpoints:[charge_state]]
2025/01/31 12:50:41 DEBU get: charge_state
2025/01/31 12:50:43 INFO successfully executed command=vehicle_data body=map[endpoints:[charge_state]]
2025/01/31 12:50:43 INFO sending command=vehicle_data body=map[endpoints:[charge_state]]
2025/01/31 12:50:43 DEBU get: charge_state
2025/01/31 12:50:44 INFO successfully executed command=vehicle_data body=map[endpoints:[charge_state]]
2025/01/31 12:50:44 INFO sending command=vehicle_data body=map[endpoints:[charge_state]]
2025/01/31 12:50:44 DEBU get: charge_state
2025/01/31 12:50:46 INFO successfully executed command=vehicle_data body=map[endpoints:[charge_state]]
2025/01/31 12:50:46 INFO received command=set_charging_amps body=map[charging_amps:4]
2025/01/31 12:50:46 INFO sending command=set_charging_amps body=map[charging_amps:4]
2025/01/31 12:50:47 INFO successfully executed command=set_charging_amps body=map[charging_amps:4]
2025/01/31 12:50:50 INFO sending command=vehicle_data body=map[endpoints:[charge_state]]
2025/01/31 12:50:50 DEBU get: charge_state
2025/01/31 12:50:51 INFO successfully executed command=vehicle_data body=map[endpoints:[charge_state]]
2025/01/31 12:50:51 INFO sending command=vehicle_data body=map[endpoints:[charge_state]]
2025/01/31 12:50:51 DEBU get: charge_state
2025/01/31 12:50:53 INFO successfully executed command=vehicle_data body=map[endpoints:[charge_state]]
2025/01/31 12:50:53 INFO sending command=vehicle_data body=map[endpoints:[charge_state]]
2025/01/31 12:50:53 DEBU get: charge_state
2025/01/31 12:50:54 INFO successfully executed command=vehicle_data body=map[endpoints:[charge_state]]
2025/01/31 12:50:54 INFO received command=charge_stop body=map[]
2025/01/31 12:50:54 INFO sending command=charge_stop body=map[]
2025/01/31 12:50:55 INFO successfully executed command=charge_stop body=map[]

Lenart12, 2025-01-31:
@Trunks1982 I added new version of addon to fix this. You can update :^)

Trunks1982, 2025-01-31:
Super das stoppen hat jetzt funktioniert.
entschuldige wenn ich dich nerve:-)

Ich hab aber immer noch diese Abbrüche:

2025/01/31 13:58:30 INFO received command=set_charging_amps body=map[charging_amps:5]
2025/01/31 13:58:30 INFO connecting to Vehicle ...
2025/01/31 13:58:30 WARN failed to scan for vehicle: ble: failed to scan for XP7----: skt: can't read hci socket: broken pipe
2025/01/31 13:58:30 INFO retrying in 3 seconds
2025/01/31 13:58:33 WARN failed to scan for vehicle: ble: failed to scan for XP7--------: skt: can't read hci socket: broken pipe
2025/01/31 13:58:33 INFO retrying in 6 seconds
2025/01/31 13:58:39 ERRO stop retrying after 3 attempts error="failed to scan for vehicle: ble: failed to scan for XP7YGCES9RB368880: skt: can't read hci socket: broken pipe"
2025/01/31 13:58:39 ERRO can't connect to vehicle error="failed to scan for vehicle: ble: failed to scan for XP7YGCES9RB368880: skt: can't read hci socket: broken pipe"

Da ich ja HA in einer VM nutze kann ich auch sehn was mir über HCI0 und HCI1 ausgegeben wird. Hier noch nen Bild aus Proxmox.

[image]

Lenart12, 2025-01-31:
Hmm hard to say... If you are using Full access version of the addon I would say that the issue is inside tesla vehicle-command project, or actually (go/ble) https://github.com/teslamotors/vehicle-command/issues/261

Trunks1982, 2025-01-31:
[image]

Würde ich auch sagen. Zb nutze ich ein Yuki was auch über Bluetooth direkt angesprochen wird und hier bricht die Verbindung nicht ab. Hier mal ein kleiner Auszug. Weiß natürlich nicht ob das weiterhilft weil er python nutzt.

[image]

Trunks1982, 2025-01-31:
[image]

Oh HCi1

Trunks1982, 2025-02-01:
Sorry das ich dich nerve aber mittlerweile stürzt hci nach dem ersten Befehl ab.

2025/02/01 10:37:36 DEBU waiting for command
2025/02/01 10:37:52 DEBU Received message topic=tb2m/XP7YGCES9RB368880/wakeup_car/set message=PRESS
2025/02/01 10:37:52 INFO Handling command key=PRESS action=wake_up body=""
2025/02/01 10:37:52 DEBU Getting proxy response url="http://localhost:5667/api/1/vehicles/XP7xxxxxxxxxxx/wake_up?wait=true"
2025/02/01 10:37:52 DEBU Canceling publish handler=tb2m_XP7xxxxxxx
2025/02/01 10:37:52 DEBU received handler=VehicleEndpoint method=POST url="/api/1/vehicles/XP7YGCES9RB368880/wake_up?wait=true" from=[::1]:47770
2025/02/01 10:37:52 INFO received command=wake_up body=map[]
2025/02/01 10:37:52 INFO connecting to Vehicle ...
2025/02/01 10:37:52 DEBU trying connecting to vehicle attempt=1
2025/02/01 10:37:52 DEBU connecting to vehicle (A)...
2025/02/01 10:37:52 DEBU scan for vehicle ...
2025/02/01 10:37:52 WARN failed to scan for vehicle: ble: failed to scan for XP7xxxxxx: skt: can't read hci socket: broken pipe
2025/02/01 10:37:52 INFO retrying in 3 seconds
2025/02/01 10:37:55 DEBU trying connecting to vehicle attempt=2
2025/02/01 10:37:55 DEBU connecting to vehicle (A)...
2025/02/01 10:37:55 DEBU scan for vehicle ...
2025/02/01 10:37:55 WARN failed to scan for vehicle: ble: failed to scan for XP7Yxxxxxxxx: skt: can't read hci socket: broken pipe
2025/02/01 10:37:55 INFO retrying in 6 seconds
2025/02/01 10:38:01 DEBU trying connecting to vehicle attempt=3

Lenart12, 2025-02-01:
This was broken by bad configuration on my side 🤦. You can update addon to tb2m-fa:0.0.16 or tbhp-fa:0.0.4. Thank you for the report

Trunks1982, 2025-02-01:
Kein problem bin auch so nen bisschen am forschen. 
zb wurde mir die docker copmose so empfohlen ka ob das was ändert. Versuche halt mit meinen Mitteln zu helfen.

services:
  tesla-ble-http-proxy:
    image: wimaha/tesla-ble-http-proxy:latest
    container_name: tesla-ble-http-proxy
    volumes:
      - ~/TeslaBleHttpProxy/key:/key
      - /var/run/dbus:/var/run/dbus
    restart: unless-stopped
    privileged: true
    network_mode: host
    cap_add:
      - NET_ADMIN
      - SYS_ADMIN
    environment:  # optional
      logLevel: debug
      httpListenAddress: ":8080"
      cacheMaxAge: "180"
      DBUS_SYSTEM_BUS_ADDRESS: "unix:path=/var/run/dbus/system_bus_socket"

Trunks1982, 2025-02-01:
es ist ein Fehler im neuen Addon:

[11:13:32] INFO: Starting TeslaBle2Mqtt v0.0.16-fa
[11:13:32] INFO: VINs: XP7YGCES9RB368880
[11:13:32] INFO: Slug: 0e0de07f_tb2m-full-access
[11:13:32] INFO: Configuration url: homeassistant://hassio/ingress/0e0de07f_tb2m-full-access
[11:13:32] INFO: Ingress url: /api/hassio_ingress/6f5L9TfvVC6RLZ6iVhUQTYC9zDgk-bBHLvjYSPumNAw/
[11:13:32] INFO: Ingress ent: /api/hassio_ingress/6f5L9TfvVC6RLZ6iVhUQTYC9zDgk-bBHLvjYSPumNAw
[11:13:32] INFO: Repo: 0e0de07f
[11:13:32] INFO: Hostname: 0e0de07f-tb2m-full-access
2025/02/01 11:13:32 INFO TeslaBleHttpProxy *undefined* is loading ...
2025/02/01 11:13:32 DEBU privateKeyFile loaded
2025/02/01 11:13:32 INFO BleControl initialized
2025/02/01 11:13:32 INFO TeslaBleHttpProxy is running!
2025/02/01 11:13:32 INFO Starting TeslaBle2Mqtt
2025/02/01 11:13:32 DEBU Running with settings="&{Vins:[XP7YGCES9RB368880] ProxyHost:http://localhost:5667 PollInterval:120 PollIntervalCharging:120 MqttHost:localhost MqttPort:1883 MqttUser:cb MqttPass:christoph82 MqttQos:0 DiscoveryPrefix:homeassistant MqttPrefix:tb2m ResetDiscovery:true SensorsYaml: LogLevel:DEBUG MqttDebug:false ReportedVersion:0.0.16-fa ReportedConfigUrl:homeassistant://hassio/ingress/0e0de07f_tb2m-full-access}"
2025/02/01 11:13:32 DEBU Using default sensors configuration
2025/02/01 11:13:32 FATA Failed to get discovery error="yaml: line 498: did not find expected key"
s6-rc: info: service legacy-services: stopping
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped

Lenart12, 2025-02-01:
New version tb2m:0.0.17. I should really test even the most "simple" patches 🤦

Trunks1982, 2025-02-01:
Danke hab ich gemacht.

Hab aber immer noch den Fehler wenn ich zb über ha was änder.

2025/02/01 11:24:45 ERRO can't connect to vehicle error="failed to scan for vehicle: ble: failed to scan for XP7YGCES9RB368880: skt: can't read hci socket: broken pipe"
2025/02/01 11:24:45 DEBU connecting to Vehicle done
2025/02/01 11:24:45 DEBU response command=set_charging_amps status=503 result=false reason="failed to scan for vehicle: ble: failed to scan for xxxxxxx: skt: can't read hci socket: broken pipe"
2025/02/01 11:24:45 ERRO Failed to handle command error="command failed: failed to scan for vehicle: ble: failed to scan for XP7YGCES9RB368880: skt: can't read hci socket: broken pipe"
2025/02/01 11:24:45 DEBU Resuming publish handler=tb2m_XP7xxxxx
2025/02/01 11:24:45 DEBU Publish state loop handler=per_vehicle for=tb2m_XP7xxxxxx
2025/02/01 11:24:45 DEBU Getting proxy response url=http://localhost:5667/api/proxy/1/vehicles/XP7Yxxxxxxx/connection_status
2025/02/01 11:24:45 DEBU received handler=ProxyCommand method=GET url=/api/proxy/1/vehicles/XP7YGCES9RB368880/connection_status from=[::1]:58108
2025/02/01 11:24:45 INFO received command=connection_status
2025/02/01 11:24:46 DEBU waiting for command
2025/02/01 11:24:46 DEBU response command=connection_status status=503 result=false reason="ble: failed to scan for XP7YGCES9RB368880: skt: can't read hci socket: broken pipe"
2025/02/01 11:24:46 DEBU Failed to get state error="failed to get connection status: command failed: ble: failed to scan for XP7YGCES9RB368880: skt: can't read hci socket: broken pipe"
2025/02/01 11:24:46 DEBU Waiting for next poll interval interval=120
2025/02/01 11:24:47 DEBU waiting for command
2025/02/01 11:25:20 DEBU Publish state loop handler=handler for=tb2m
2025/02/01 11:25:20 INFO Publishing topic=tb2m/tb2m/uptime/state access path=uptime state=240 old_state=120
2025/02/01 11:25:20 DEBU Waiting for next poll interval interval=120

Lenart12, 2025-02-01:
[image] Do you see such badge on your install?

Trunks1982, 2025-02-01:
> [image] Do you see such badge on your install?
ja.

Lenart12, 2025-02-01:
Hmmm. Maybe you need to uninstall and reinstall, not just update 🤔

[30 more replies]