wimaha/TeslaBleHttpProxy #86: Missing fields on dashboard

MrBLJ, 2025-01-13:
I'm sorry if this sound very dumb, but I'm faced with a peculiar issue : I'm running the container using the docker comfiguration specified in the documentation (% logLevel and TZ) :

```yaml
services:
  tesla-ble-http-proxy:
    image: wimaha/tesla-ble-http-proxy
    container_name: TeslaBleHttpProxy
    environment:
      - cacheMaxAge=30
      - logLevel=debug
      - TZ=Europe/Paris
    volumes:
      - ~/TeslaBleHttpProxy/key:/key
      - /var/run/dbus:/var/run/dbus
    restart: always
    privileged: true
    network_mode: host
    cap_add:
      - NET_ADMIN
      - SYS_ADMIN
```

I can access the Web UI, but it's missing fields :
[image]

Logs are ok, I think, in portainer :

```
2025/01/13 15:51:45 INFO TeslaBleHttpProxy is loading ...
2025/01/13 15:51:45 DEBU LogLevel set to debug
2025/01/13 15:51:45 INFO TeslaBleHttpProxy is running!
```

Same using logs from cli :

```
root@DietPi:~/TeslaBleHttpProxy# docker logs --since 12h TeslaBleHttpProxy
2025/01/13 15:51:45 INFO TeslaBleHttpProxy is loading ...
2025/01/13 15:51:45 DEBU LogLevel set to debug
2025/01/13 15:51:45 INFO TeslaBleHttpProxy is running!
```

I tried recreating the container a few time, with system prune in between. Removing TZ and loglevel does not change anything.

I also tried using different browsers (firefox, brave, edge, safari on iOs) in regular and private browsing (0 extension). Same result.

The only thing I could find is a mention of a missing favicons, in my browser console's logs :

` GET http://192.168.20.149:8080/favicon.ico 404 (Not Found)`

I'm sorry to ask that but : Am I missing something obvious here ?

wimaha, 2025-01-13:
Hi this is a bit strange, because the logs are not complete.
Should look like this:

```
2025/01/13 16:55:06 INFO TeslaBleHttpProxy *undefined* is loading ...
2025/01/13 16:55:06 DEBU LogLevel set to debug
2025/01/13 16:55:06 INFO TeslaBleHttpProxy httpListenAddress=:8080
2025/01/13 16:55:06 DEBU privateKeyFile loaded
2025/01/13 16:55:06 INFO BleControl initialized
2025/01/13 16:55:06 INFO TeslaBleHttpProxy is running!
```

So in your LogFile the version is missing, and some log entries.

Which Version are you using? Are there any errors after opening the dashboard?

MrBLJ, 2025-01-13:
Thanks for you reply !

> Which Version are you using ?
I pulled the image today for the first time, so I guess it's the latest on _main_. My compose is defined as follow : `image: wimaha/tesla-ble-http-proxy`.

> Are there any errors after opening the dashboard?
No errors whatsoever, expect for the missing favicon.  Fields are missing that's the only thing I can confirm.  As for the logs, it's different than what you posted though.

MrBLJ, 2025-01-13:
(in)sanity check : problem solved, after the third container rebuild :

```bash
docker compose stop tesla-ble-http-proxy
docker compose rm -f
docker compose pull
docker compose up --build -d
```

Docker logs :
```
root@DietPi:~/TeslaBleHttpProxy# docker compose stop tesla-ble-http-proxy
[+] Stopping 1/1
 ✔ Container TeslaBleHttpProxy  Stopped                                    6.7s
root@DietPi:~/TeslaBleHttpProxy# docker compose rm -f
Going to remove TeslaBleHttpProxy
[+] Removing 1/0
 ✔ Container TeslaBleHttpProxy  Removed                                    0.0s
root@DietPi:~/TeslaBleHttpProxy# docker compose pull
[+] Pulling 4/4
 ✔ tesla-ble-http-proxy Pulled                                            25.4s
   ✔ 1e9fa5e12071 Pull complete                                            3.8s
   ✔ cd0bfe1a5b6f Pull complete                                           10.3s
   ✔ f04d23a2fd34 Pull complete                                           10.5s
root@DietPi:~/TeslaBleHttpProxy# docker compose up --build -d
[+] Running 1/1
 ✔ Container TeslaBleHttpProxy  Started 
```

Container logs :
```
2025/01/13 19:15:59 INFO TeslaBleHttpProxy 2.0.3 is loading ...
2025/01/13 19:15:59 DEBU LogLevel set to debug
2025/01/13 19:15:59 INFO TeslaBleHttpProxy httpListenAddress=:8080
2025/01/13 19:15:59 ERRO failed to load private key. err="open key/private.pem: no such file or directory"
2025/01/13 19:15:59 WARN BleControl could not be initialized!
2025/01/13 19:15:59 INFO TeslaBleHttpProxy is running!
```

UI looks fine:
[image]

Still an error about the favicon though ^^.

Sorry for the disturbance.

closed 2025-01-13 by MrBLJ

wimaha, 2025-01-13:
Thanks for the feedback. The favicon is missing. 😃