wimaha/TeslaBleHttpProxy #96: Initial connection working after pair, however ever since driving away and come back, no connectivity.

jkerekes99, 2025-01-28:
As per title, after the initial bluetooth pair with car, everything was working perfectly with EVCC.

However after driving away and coming back, car no longer connects to proxy.
Here are logs:

VIN number has been replaced with the text "obfuscated".

2025/01/26 07:55:16 INFO connecting to Vehicle ...
2025/01/26 07:55:32 WARN failed to connect to vehicle (A): failed to find BLE beacon for obfuscated (S56123af55c4fc242C): can't scan: received scan response c0:95:6d:50:e6:42 with no associated Advertising Data packet
2025/01/26 07:55:32 INFO retrying in 3 seconds
2025/01/26 07:55:53 WARN failed to connect to vehicle (A): failed to find BLE beacon for obfuscated (S56123af55c4fc242C): can't scan: received scan response c0:95:6d:50:e6:42 with no associated Advertising Data packet
2025/01/26 07:55:53 INFO retrying in 6 seconds
2025/01/26 07:56:15 ERRO stop retrying after 3 attempts error="failed to connect to vehicle (A): failed to find BLE beacon for obfuscated (S56123af55c4fc242C): can't scan: received scan response c0:95:6d:50:e6:42 with no associated Advertising Data packet"

This repeats continuely.

I rebooted my raspberry pi and now it doesn't even attempt to connect, all I get in the logs is:
2025/01/28 12:44:38 INFO TeslaBleHttpProxy 2.0.3 is loading ...
2025/01/28 12:44:38 INFO TeslaBleHttpProxy httpListenAddress=:8080
2025/01/28 12:44:38 INFO BleControl initialized
2025/01/28 12:44:38 INFO TeslaBleHttpProxy is running!

I dont know if I can even force a reconnect?
Where do I go from here?

Is there a command I can run to see the status of the connection to the car?

Is the way to force a reconnect?

Trunks1982, 2025-01-28:
Diese Probleme habe ich auch. Mit einem Auto funktioniert das solala. Mit dem zweiten so gut wie gar nicht.

jkerekes99, 2025-01-30:
I managed to do a charge session today, however, it only charged the car to 79% even though I set the charge limit to 80% in EVCC.

While it was charging the Tesla app always showed 1% higher than EVCC.

EVCC log showed:
[lp-1  ] WARN 2025/01/30 10:44:36 vehicle soc: unexpected status: 503 (Service Unavailable) (ignored by estimator)
[lp-1  ] ERROR 2025/01/30 10:44:36 vehicle soc limit: unexpected status: 503 (Service Unavailable)
[lp-1  ] ERROR 2025/01/30 10:44:36 vehicle range: unexpected status: 503 (Service Unavailable)
[lp-1  ] WARN 2025/01/30 10:44:46 vehicle soc: unexpected status: 503 (Service Unavailable) (ignored by estimator)
[lp-1  ] ERROR 2025/01/30 10:44:46 vehicle soc limit: unexpected status: 503 (Service Unavailable)
[lp-1  ] ERROR 2025/01/30 10:44:46 vehicle range: unexpected status: 503 (Service Unavailable)
[lp-1  ] WARN 2025/01/30 13:49:36 vehicle soc: unexpected status: 503 (Service Unavailable) (ignored by estimator)
[lp-1  ] ERROR 2025/01/30 13:49:36 vehicle soc limit: unexpected status: 503 (Service Unavailable)
[lp-1  ] ERROR 2025/01/30 13:49:36 vehicle range: unexpected status: 503 (Service Unavailable)
[lp-1  ] WARN 2025/01/30 13:49:46 vehicle soc: unexpected status: 503 (Service Unavailable) (ignored by estimator)
[lp-1  ] ERROR 2025/01/30 13:49:46 vehicle soc limit: unexpected status: 503 (Service Unavailable)
[lp-1  ] ERROR 2025/01/30 13:49:46 vehicle range: unexpected status: 503 (Service Unavailable)

The docker logs for TeslaBleHttpProxy did not show any obvious errors.

Trunks1982, 2025-02-01:
> Ich habe es heute geschafft, eine Ladesitzung zu machen, aber es hat das Auto nur auf 79 % aufgeladen, obwohl ich das Ladelimit in EVCC auf 80 % eingestellt habe.
> 
> Während des Aufladens zeigte die Tesla-App immer 1% mehr als EVCC.
> 
> Das EVCC-Protokoll zeigte:[lp-1 ] WARNUNG 2025/01/30 10:44:36 Fahrzeug soc: unerwarteter Status: 503 (Dienst nicht verfügbar) (vom Schätzer ignoriert)[lp-1 ] FEHLER 2025/01/30 10:44:36 Fahrzeug soc-Limit: unerwarteter Status: 503 (Service nicht verfügbar)[lp-1 ] FEHLER 2025/01/30 10:44:36 Fahrzeugreichweite: unerwarteter Status: 503 (Service nicht verfügbar)[lp-1 ] WARNUNG 2025/01/30 10:44:46 Fahrzeug soc: unerwarteter Status: 503 (Service nicht verfügbar) (vom Schätzer ignoriert)[lp-1 ] FEHLER 2025/01/30 10:44:46 Fahrzeug-Soc-Limit: unerwarteter Status: 503 (Service nicht verfügbar)[lp-1 ] FEHLER 2025/01/30 10:44:46 Fahrzeugreichweite: unerwarteter Status: 503 (Service nicht verfügbar)[lp-1 ] WARNUNG 2025/01/30 13:49:36 Fahrzeug soc: unerwarteter Status: 503 (Service nicht verfügbar) (vom Schätzer ignoriert)[lp-1 ] FEHLER 2025/01/30 13:49:36 Fahrzeug-Soc-Limit: unerwarteter Status: 503 (Service nicht verfügbar)[lp-1 ] FEHLER 2025/01/30 13:49:36 Fahrzeugreichweite: unerwarteter Status: 503 (Service nicht verfügbar)[lp-1 ] WARNUNG 2025/01/30 13:49:46 Fahrzeug soc: unerwarteter Status: 503 (Service nicht verfügbar) (vom Schätzer ignoriert)[lp-1 ] FEHLER 2025/01/30 13:49:46 Fahrzeug-Soc-Limit: unerwarteter Status: 503 (Service nicht verfügbar)[lp-1 ] FEHLER 2025/01/30 13:49:46 Fahrzeugreichweite: unerwarteter Status: 503 (Service nicht verfügbar)
> 
> Die Docker-Protokolle für TeslaBleHttpProxy zeigten keine offensichtlichen Fehler.

Läuft es bei dir noch zuverlässig?
Benutzt du einen usb dongle?

Trunks1982, 2025-02-03:
So ich hab noch mal nen test gemacht und nur noch HA auf dem Rechner installiert ohne irgendwas dazwischen und es klappt genauso nicht. Ich würde sagen man muss wie oben schon erwähnt das gerät direkt ansprechen.

jkerekes99, 2025-02-05:
> Is it still working reliably for you?
> Are you using a USB dongle?

So its working now semi-reliably in that so far after rebooting the Pi and updating everything with apt-get upgrade / update its worked every single time so far.

The only issue is that it does get out of sync with EVCC where it will only charge until 79% even if I have it set to 80%.

And when initially plugging the car into the charger I get this:
[image]

Trunks1982, 2025-02-05:
Bei mir ist es eher so das evcc 1-2% schneller ist.
Aber du kannst gerne mal diese compose probieren. Hier wird das Bluetooth direkt angesprochen und nicht über Umwege wie dbus. Bei USB Dongle definitiv die bessere Wahl. Aber vielleicht auch besser für den PI.

services:
  tesla-ble-http-proxy:
    image: wimaha/tesla-ble-http-proxy:latest
    container_name: tesla-ble-http-proxy
    volumes:
      - ~/TeslaBleHttpProxy/key:/key
    restart: always
    privileged: true
    network_mode: host
    cap_add:
      - NET_ADMIN
      - SYS_ADMIN
    devices:
      - "/dev/hci0:/dev/hci0"  # Direktzugriff auf den Bluetooth-Adapter
    environment: #optional
      logLevel: "debug"
      httpListenAddress: ":9191"
      cacheMaxAge: 240

jkerekes99, 2025-06-03:
It's still an issue, but if EVCC is deviating from the proxies value, then I would say the issue is with EVCC not the proxy.
So this can be closed if needed.

closed 2025-06-03 by jkerekes99