wimaha/TeslaBleHttpProxy #98: failed to connect to vehicle (A): context deadline exceeded

pre5ton, 2025-02-05:
Hello,

Hoping someone can point me towards a solution.  Have a new Raspberry Pi 5 located immediately next to the vehicle (~3 feet away), but continue to receive the error "failed to connect to vehicle (A): context deadline exceeded".  Followed documentation exactly as outlined. Any support is appreciated. 

I went through possible issues and cannot identify what I'm doing wrong, list of checks below.

Used VNC Viewer to access Raspberry Pi GUI to ensure bluetooth is enabled.
Ensured vehicle is awake (door open, screen on)
No errors in logs
Multiple positions of Raspberry Pi

Logs

2025/02/05 02:58:47 INFO TeslaBleHttpProxy 2.0.3 is loading ...
2025/02/05 02:58:47 INFO TeslaBleHttpProxy httpListenAddress=:8080
2025/02/05 02:58:47 INFO BleControl initialized
2025/02/05 02:58:47 INFO TeslaBleHttpProxy is running!
2025/02/05 03:02:50 can't accept: listner timed out

michael-robbins, 2025-02-16:
How'd you setup the container/software? Where did you get up to? Did you setup the virtual key thing?

pre5ton, 2025-02-16:
Followed instructions outlined here exactly - https://github.com/wimaha/TeslaBleHttpProxy/blob/main/docs/installation.md

Software is running in docker.

Can access dashboard page and created the virtual keys without issue. 

Error is received when attempting to send the virtual key to the vehicle.

michael-robbins, 2025-02-17:
Ah yeah, I've got a rpi5 with it running in docker myself, same setup, and it's working, so just trying to find what's different with your setup.

Can you paste here the docker run command, or docker compose config?

pre5ton, 2025-02-18:
Here is the docker compose config:

`services:
  tesla-ble-http-proxy:
    image: wimaha/tesla-ble-http-proxy
    container_name: tesla-ble-http-proxy
    environment:
      - cacheMaxAge=30 # Optional, but recommended to set this to anything more>
    volumes:
      - ~/TeslaBleHttpProxy/key:/key
      - /var/run/dbus:/var/run/dbus
    restart: always
    privileged: true
    network_mode: host
    cap_add:
      - NET_ADMIN
      - SYS_ADMIN`

Running the below OS version loaded using the official Raspberry Pi imager.

Raspberry Pi OS with desktop
Release date: November 19th 2024
System: 64-bit
Kernel version: 6.6
Debian version: 12 (bookworm)
Size: 1,179MB

pre5ton, 2025-02-18:
I tried to send the keys to the vehicle again while I had the Pi running and this time it worked!  Not sure what changed...

I have two vehicles (both Teslas) so setup the key in both vehicles.  The first one I sent the keys to the vehicle when I was not in the car and it says they were sent, went to the garage and there was nothing on the display, but tapped the key card anyways.  Locks show a "BLE" key now.  Second one I sent the key while in the vehicle, tapped the card and it added "unknown key". 

Now when setting up in EVCC I'm receiving the below errors when trying to retrieve vehicle SOC or sending charging commands.  Effectively the same errors as in the subject post.

Proxy is at 192.168.1.120 and running on port 8080.

[main ] INFO 2025/02/17 17:53:26 evcc 0.200.1
[main ] INFO 2025/02/17 17:53:26 using config file: /etc/evcc.yaml
[db ] INFO 2025/02/17 17:53:26 using sqlite database: /root/.evcc/evcc.db
[main ] INFO 2025/02/17 17:53:27 listening at :7070
[site ] INFO 2025/02/17 17:53:27 site config:
[site ] INFO 2025/02/17 17:53:27 meters: grid ✓ pv ✓ battery ✗
[site ] INFO 2025/02/17 17:53:27 grid: power ✓ energy ✗ currents ✓
[site ] INFO 2025/02/17 17:53:27 pv 1: power ✓ energy ✓ currents ✓
[site ] INFO 2025/02/17 17:53:27 vehicles:
[site ] INFO 2025/02/17 17:53:27 vehicle 1: range ✓ finish ✗ status ✓ climate ✗ wakeup ✓
[site ] INFO 2025/02/17 17:53:27 vehicle 2: range ✓ finish ✗ status ✓ climate ✗ wakeup ✓
[lp-1 ] INFO 2025/02/17 17:53:27 loadpoint 1:
[lp-1 ] INFO 2025/02/17 17:53:27 mode: pv
[lp-1 ] INFO 2025/02/17 17:53:27 charger: power ✓ energy ✗ currents ✓ phases ✗ wakeup ✗
[lp-1 ] INFO 2025/02/17 17:53:27 meters: charge ✓
[lp-1 ] INFO 2025/02/17 17:53:27 charge: power ✓ energy ✗ currents ✓
[lp-1 ] DEBUG 2025/02/17 17:53:27 phase timer inactive
[lp-1 ] DEBUG 2025/02/17 17:53:27 pv timer inactive
[site ] DEBUG 2025/02/17 17:53:27 ----
[lp-1 ] DEBUG 2025/02/17 17:53:27 charge power: 0W
[lp-1 ] DEBUG 2025/02/17 17:53:27 charge currents: [0 0 0]A
[site ] DEBUG 2025/02/17 17:53:27 grid power: 610W
[site ] DEBUG 2025/02/17 17:53:31 pv 1 power: 293W
[site ] DEBUG 2025/02/17 17:53:48 grid currents: [4.57 3.92 0]A
[site ] DEBUG 2025/02/17 17:53:48 site power: 610W
[lp-1 ] DEBUG 2025/02/17 17:53:48 charge voltages: [2.1 2.1 2.1]V
[lp-1 ] DEBUG 2025/02/17 17:53:48 charger status: B
[lp-1 ] INFO 2025/02/17 17:53:48 car connected
[lp-1 ] DEBUG 2025/02/17 17:53:48 vehicle api refresh
[lp-1 ] DEBUG 2025/02/17 17:53:48 pv timer elapse
[lp-1 ] DEBUG 2025/02/17 17:53:48 pv timer inactive
[main ] ERROR 2025/02/17 17:54:18 vehicle status: Get "http://192.168.1.120:8080/api/1/vehicles/5YJ3E1EA7MXXXXX63/vehicle_data?endpoints=charge_state": context deadline exceeded (Client.Timeout exceeded while awaiting headers)
[main ] ERROR 2025/02/17 17:54:48 vehicle status: Get "http://192.168.1.120:8080/api/1/vehicles/5YJ3E1EAXMXXXXX22/vehicle_data?endpoints=charge_state": context deadline exceeded (Client.Timeout exceeded while awaiting headers)
[lp-1 ] INFO 2025/02/17 17:54:48 vehicle updated: unknown -> Jamie
[lp-1 ] INFO 2025/02/17 17:54:48 vehicle updated: Jamie -> unknown
[lp-1 ] DEBUG 2025/02/17 17:54:48 pv charge current: 0A = 0A + -2.65A (610W @ 1p)
[lp-1 ] DEBUG 2025/02/17 17:54:48 pv enable timer reset
[lp-1 ] DEBUG 2025/02/17 17:54:48 pv timer inactive
[lp-1 ] INFO 2025/02/17 17:54:48 vehicle updated: unknown -> Jamie
[lp-1 ] DEBUG 2025/02/17 17:56:52 set charge mode: now
[site ] DEBUG 2025/02/17 17:56:52 ----
[lp-1 ] DEBUG 2025/02/17 17:56:52 charge power: 0W
[lp-1 ] DEBUG 2025/02/17 17:56:52 charge currents: [0 0 0]A
[site ] DEBUG 2025/02/17 17:56:55 pv 1 power: 278W
[site ] DEBUG 2025/02/17 17:56:58 grid power: 534W
[site ] DEBUG 2025/02/17 17:57:12 grid currents: [3.99 4.07 0]A
[site ] DEBUG 2025/02/17 17:57:12 site power: 534W
[lp-1 ] DEBUG 2025/02/17 17:57:12 charge voltages: [2.1 2.1 2.1]V
[lp-1 ] DEBUG 2025/02/17 17:57:12 charger status: B
[lp-1 ] ERROR 2025/02/17 17:57:42 vehicle soc: Get "http://192.168.1.120:8080/api/1/vehicles/5YJ3E1EA7MXXXXX63/vehicle_data?endpoints=charge_state": context deadline exceeded (Client.Timeout exceeded while awaiting headers)
[lp-1 ] DEBUG 2025/02/17 17:57:42 max charge current: 32A
[lp-1 ] DEBUG 2025/02/17 17:57:42 charger enable
[lp-1 ] DEBUG 2025/02/17 17:57:42 wake-up timer: start

Here is my EVCC.yaml:

# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

[... 31 lines cut ...]
    type: template
    template: tesla-ble
    title: Jamie
    capacity: 50
    vin: 5YJ3E1EA7MXXXXX63
    url: http://192.168.1.120
    port: 8080
  - name: preston_tesla
    type: template
    template: tesla-ble
    title: Preston
    capacity: 50
    vin: 5YJ3E1EAXMXXXXX22
    url: http://192.168.1.120
    port: 8080

loadpoints:
- title: Home
  charger: wallbox4
  mode: pv

site:
  title: Home
  meters:
    grid: grid1

pre5ton, 2025-02-18:
Still testing, but believe I may have solved it.  The issue stems around the queries failing to resolve due to the DNS being used.  I am not using my own router and am using my AT&T Fiber combined modem/router device which uses AT&T's DNS to resolve queries.  It was odd to me that both the raspberry pi 5 running TeslaBleHttpProxy and my unraid server running EVCC were both returning similar errors. I changed both my unraid server and my pi to use Google & CloudFlare DNS servers and the issue seems to be resolved.

michael-robbins, 2025-02-19:
Given you're talking to the IP address of the raspberry pi, DNS shouldn't even be in the picture here, and nothing requires internet/hostname resolution either, unless I missed something above.

Try to post things inside code block (with 3x backticks ` ` ` in a row without spaces before and after):
```
like this

easy to paste

formats like code
```

My setup is similar:
```
  tesla-ble-http-proxy:
    image: wimaha/tesla-ble-http-proxy:latest
    container_name: tesla-ble-http-proxy

    privileged: true
    network_mode: host
    cap_add:
      - NET_ADMIN
      - SYS_ADMIN

    environment:
      - httpListenAddress=:8081

    volumes:
      - tesla-ble-http-proxy-key:/key
      - /var/run/dbus:/var/run/dbus

    restart: unless-stopped
```

Along with a bog standard Raspberry Pi 4 running in the garage alongside the car.

Key transmission 'just worked', sometimes I'll see errors around `the vehicle is already connected to the maximum number of BLE devices` but I think that might be the bluetooth on the raspberry pi acting weirdly... not exactly sure...

pre5ton, 2025-02-20:
My current issue is the same one as in the original post.  I can't get TeslaBleHttpProxy to send the keys to my vehicle (errors provided above).  

Also, corrected formatting of above posts.

pre5ton, 2025-03-18:
Providing an update and closing as I have resolved the issue, but I'm not certain it was the problem.

My wife and I both have multiple phones that are all connected to the vehicles and I noticed there were 3-4 keys active at any time when the cars were in the garage.  I turned off bluetooth on all of our phones so that the virtual keys would disconnect from the vehicles.  When I did this and then used TeslaBLEHTTPProxy to send the keys to the vehicle, it worked.

closed 2025-03-18 by pre5ton