wimaha/TeslaBleHttpProxy #101: evcc custom charger

bill340, 2025-02-17:
Is there any chance to get an evcc custom charger that is used to charge as long as there is a bluetooth connection to the car?

wimaha, 2025-03-31:
I don't understand ... you mean that the car should considered plugged a long as a ble connection can be establish?

wimaha, 2025-06-01:
Using Bluetooth connection status to trigger charging in EVCC isn’t reliable, as it only indicates the Tesla is nearby (~5-10m), not plugged into the TWC3. A better solution is to add a manual “Vehicle Plugged In” toggle to the TeslaBleHttpProxy web interface (http://YOUR_IP:8080/dashboard). This could set a flag in the proxy that EVCC reads via its API to enable charging.