wimaha/TeslaBleHttpProxy #102: context deadline exceeded on charge_state endpoint ONLY

tomaszrondio, 2025-02-20:
Today I had an issue `vehicle soc: Get .../vehicle_data?endpoints=charge_state": context deadline exceeded (Client.Timeout exceeded while awaiting headers)` and interesting thing is that only this endpoint was timing out. Other I tested (door lock, unlock etc) worked fine. Docker logs show nothing. Where do I look?

pre5ton, 2025-03-19:
I have the same problem.  Seems like it will work without any errors for hours and then will start throwing this error on vehicle soc and vehicle range on every refresh.

hanzwo, 2025-03-23:
I have / had the same problem. My workaround for now is that I restart the container every time I unplug the car. This works for now.

pre5ton, 2025-03-25:
> I have / had the same problem. My workaround for now is that I restart the container every time I unplug the car. This works for now.

I tried to do this and I still receive the error even after restarting the container each time...

wimaha, 2025-03-31:
For me it is working fine. 
`context deadline exceeded` is probably because the ble connection isn't working...

pre5ton, 2025-04-01:
I believe I identified the problem & solution.  Based on the service mode screens, I think the Tesla Model 3 can only handle having 3 keys connected at a time.  When my vehicles are parked in the garage and my spouse and I are at home, 3 keys will be continually connected (3 iPhones).  I noticed on the "Locks" screen of each vehicle that it would show these 3 phones connected and as active keys, but not the BLE from my server.  I decided to remove one of the iPhones from being an active key completely so that only 2 phones would regularly connect.  This resulted in the BLE proxy from my server immediately showing as an active key and the resulting errors went away. 

Theorizing that this was the same issue that I encountered when originally trying to setup the proxy and send the keys to the vehicle.  The cars wouldn't accept a new key being sent from the proxy when 3 keys were already connected which resulted in timeouts (context deadline exceeded).

Zzzzz350, 2025-04-14:
I have the same problem. I don't think that this is the key number because in my case I have only 3 keys (my phone, a tesla key and this proxy).
The proxy worked all the day and the next day, I receive the message "context deadline exceeded". 
I use it to manage my solar charge so I need that this proxy is available all the time. I will try to reboot the raspberry with this proxy every night...

Zzzzz350, 2025-04-14:
I forget the most important : Thanks a lot Wimaha for this proxy

wimaha, 2025-06-01:
I saw the context deadline exceeded on charge_state endpoint **ONLY**. I think it is because evcc requests the endpoint for soc, max soc and range three times in a row.
I will try if the cache feature could help and configured 5s cache. You can also try with the newest version.

wimaha, 2025-06-03:
The new version is live on docker.

Zzzzz350, 2025-06-04:
Thanks Wimaha.
The new version 2.1.2 is intalled here.
I need one or two week to be sure that problem is fixed.
I will give you my feedback in one or two week