wimaha/TeslaBleHttpProxy #107: Charging current in amperes not possible with one decimal place

HAL-US, 2025-04-08:
This works :

httpPost('http://lochist:8080/api/1/vehicles/(vin)/command/set_charging_amps', '{"charging_amps": "3"}', { timeout: 10000, responseType: 'text' }, async (err, response) => {

This works not:

httpPost('http://lochist:8080/api/1/vehicles/(vin)/command/set_charging_amps', '{"charging_amps": "3.8"}', { timeout: 10000, responseType: 'text' }, async (err, response) => {

To be able to control excess charging from the photovoltaic system, the charging current would be required with one decimal point.

Example chraging Ampers :
3A , 3.1A,  3.2A and so on

Could this still be included ?

wimaha, 2025-04-12:
Hi, this is not supported by the vehicle. So unfortunately we cannot include this.

closed 2025-04-12 by wimaha

HAL-US, 2025-04-12:
Is that safe? With Fleet-API, the charging current of the vehicle can be regulated in 0.1A steps.  I have verified this on my Model Y