wimaha/TeslaBleHttpProxy #124: Version 2.1.3 - Vehicle Bluetooth not in range

hanzwo, 2025-06-30:
Version 2.0.3 - works fine. If the vehicle is asleep, this version is able to wake the vehicle.

Version 2.1.3 - doesn't work, if the vehicle is asleep, this version is not able to wake the vehicle. Errors are that the vehicle is not in de Bluetooth range which is strange because if I revert to the 2.0.3 version, restart the container, the vehicle is woken up if EVCC asks. I didn't wake up the vehicle myself through the app or opening the doors.

hanzwo, 2025-06-30:
Can it be related to the new environment variable `scanTimeout`? Default value is 2 (sec?) I think? Maybe that's too short?

demod-au, 2025-07-14:
I am also getting this error.   Seems to always happen if I do a "vehicle_data" Get, quickly followed by a "body_controller_state".  

I'm independently polling every 5 min to see if car is awake and charge port is open.    Which allows the car to charge and adjust to solar.  Sometimes the poll happens right after a vehicle_data or a set_charging_amps. 

2025/07/14 10:53:38 DEBU Received HTTP request Handler=VehicleData Method=GET Endpoint="/api/1/vehicles/XXXXXXX/vehicle_data?endpoints=charge_state" Client=10.1.1.102:50688
2025/07/14 10:53:38 INFO Connecting to Vehicle ...
2025/07/14 10:53:38 DEBU Connecting to vehicle (Attempt 1) ...
2025/07/14 10:53:38 DEBU Scanning for vehicle ...
2025/07/14 10:53:38 DEBU Beacon found LocalName=XXX Address=XXX RSSI=-77
2025/07/14 10:53:39 DEBU Creating vehicle object ...
2025/07/14 10:53:39 DEBU Connecting ...
2025/07/14 10:53:39 DEBU Starting VCSEC session ...
2025/07/14 10:53:39 DEBU Car successfully wakeup
2025/07/14 10:53:39 DEBU Starting Infotainment session ...
2025/07/14 10:53:40 INFO Connection to vehicle established
2025/07/14 10:53:40 DEBU Operating connection ...
2025/07/14 10:53:40 INFO Executing command Command=vehicle_data Body=map[endpoints:[charge_state]]
2025/07/14 10:53:40 INFO Successfully executed Command=vehicle_data Body=map[endpoints:[charge_state]]
2025/07/14 10:53:40 DEBU Response Command=vehicle_data Status=200 Result=true Reason="The request was successfully processed."
2025/07/14 10:53:50 DEBU Received HTTP request Handler=BodyControllerState Method=GET Endpoint=/api/1/vehicles/XXXXXXXXXX/body_controller_state Client=10.1.1.102:50710
2025/07/14 10:53:50 DEBU Scanning for vehicle ...
2025/07/14 10:53:58 can't accept: listner timed out
2025/07/14 10:54:00 DEBU Response Command=body-controller-state Status=503 Result=false Reason="Vehicle is not in range: ble: failed to scan for XXXXXXXXX: context deadline exceeded"
2025/07/14 10:54:09 DEBU Connection timeout ...
2025/07/14 10:54:11 DEBU Waiting for next command ...

I have set  scanTimeout to 10. 

not sure if this is a bug?  or car is not responding.

Tesla234987234sdf, 2025-09-06:
I have same issue since few weeks. Also tested old sw versions. Same issue. Raspberry Pi is direct next to the car... I dont know whats happen.... Sometime it got connection. But mostly not.

wimaha, 2025-10-31:
Oh, I See that some people have problems since we first scan for the BLE-Beacon and then connect to the car. I will make an update to optionally use the old way, wich appears to be more stable for some.

wimaha, 2026-01-06:
After a bit more testing I think that increasing the scanTimeout to > 2 sec will solve the problem. I will update the readme accordingly.

closed 2026-01-06 by wimaha