wimaha/TeslaBleHttpProxy #127: "Failed to Connect to Vehicle" after connecting and starting a charge session

jkerekes99, 2025-07-16:
I don't know what to do anymore.
I've tried to get this proxy stable, however no matter what I do I'm getting issues.

Using version 2.1.3.
Performed all the raspberypi updates and rebooted.

The proxy is able to connect to the car initially when I get home (around 6pm).
Charge session doesn't start until midnight.

EVCC is able to tell the car to start charging via the proxy at midnight.
However EVCC is set to stop charging at 4am.

When I woke up this morning I found that EVCC failed to stop the charge.

docker logs show (VIN obfuscated):

2025/07/16 21:51:09 INFO Connecting to Vehicle ...
2025/07/16 21:51:09 WARN failed to connect to vehicle (A): ble: failed to scan for XXXXXXXXXXXXXXXXX: received scan response 7b:6f:9a:65:9d:76 with no associated Advertising Data packet
2025/07/16 21:51:12 WARN failed to connect to vehicle (A): ble: failed to scan for XXXXXXXXXXXXXXXXX: received scan response 7b:6f:9a:65:9d:76 with no associated Advertising Data packet
2025/07/16 21:51:18 ERRO Stop retrying after 3 attempts Error="failed to connect to vehicle (A): ble: failed to scan for XXXXXXXXXXXXXXXXX: received scan response 7b:6f:9a:65:9d:76 with no associated Advertising Data packet"
2025/07/16 21:51:18 ERRO Cannot connect to vehicle Error="failed to connect to vehicle (A): ble: failed to scan for XXXXXXXXXXXXXXXXX: received scan response 7b:6f:9a:65:9d:76 with no associated Advertising Data packet"

This repeats over and over through the night.

Command output:

jkerekes@ble-proxy:~ $ rfkill list
0: hci0: Bluetooth
	Soft blocked: no
	Hard blocked: no

jkerekes@ble-proxy:~ $ hciconfig -a
hci0:	Type: Primary  Bus: USB
	BD Address: 3C:64:CF:73:DE:EA  ACL MTU: 1021:6  SCO MTU: 255:12
	UP RUNNING
	RX bytes:243801 acl:0 sco:0 events:260 errors:0
	TX bytes:172534 acl:6174 sco:0 commands:779 errors:0
	Features: 0xff 0xff 0xff 0xfe 0xdb 0xfd 0x7b 0x87
	Packet type: DM1 DM3 DM5 DH1 DH3 DH5 HV1 HV2 HV3
	Link policy: RSWITCH HOLD SNIFF PARK
	Link mode: PERIPHERAL ACCEPT
	Name: 'TP-Link Bluetooth USB Adapter���'
Can't read class of device on hci0: Connection timed out (110)

jkerekes@ble-proxy:~ $ lsusb
Bus 001 Device 004: ID 2357:0604 TP-Link TP-Link Bluetooth USB Adapter
Bus 001 Device 003: ID 0424:ec00 Microchip Technology, Inc. (formerly SMSC) SMSC9512/9514 Fast Ethernet Adapter
Bus 001 Device 002: ID 0424:9514 Microchip Technology, Inc. (formerly SMSC) SMC9514 Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

jkerekes@ble-proxy:~ $ dmesg | grep tooth
[    4.360870] usb 1-1.4: Product: TP-Link Bluetooth USB Adapter
[   12.235043] Bluetooth: Core ver 2.22
[   12.235268] Bluetooth: HCI device and connection manager initialized
[   12.235306] Bluetooth: HCI socket layer initialized
[   12.235326] Bluetooth: L2CAP socket layer initialized
[   12.235394] Bluetooth: SCO socket layer initialized
[   12.399591] Bluetooth: hci0: RTL: examining hci_ver=0a hci_rev=000b lmp_ver=0a lmp_subver=8761
[   12.401573] Bluetooth: hci0: RTL: rom_version status=0 version=1
[   12.401624] Bluetooth: hci0: RTL: loading rtl_bt/rtl8761bu_fw.bin
[   12.418400] Bluetooth: hci0: RTL: loading rtl_bt/rtl8761bu_config.bin
[   12.420317] Bluetooth: hci0: RTL: cfg_sz 6, total sz 30210
[   12.693583] Bluetooth: hci0: RTL: fw version 0xdfc6d922
[   16.157372] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[   16.157416] Bluetooth: BNEP filters: protocol multicast
[   16.157446] Bluetooth: BNEP socket layer initialized
[   16.178836] Bluetooth: MGMT ver 1.23
[265586.616825] Bluetooth: hci0: Opcode 0x2037 failed: -22
[265588.661048] Bluetooth: hci0: command tx timeout
[595455.649554] Bluetooth: hci0: command tx timeout
[595457.729561] Bluetooth: hci0: command tx timeout
[595466.609623] Bluetooth: hci0: command tx timeout
[595468.689662] Bluetooth: hci0: command tx timeout
[595468.689743] Bluetooth: hci0: Resetting usb device.
[595468.690064] Bluetooth: hci0: urb e41ea4f2 failed to resubmit (2)
[595468.690133] Bluetooth: hci0: urb d42cf75c failed to resubmit (2)
[595468.690301] Bluetooth: hci0: urb ec39a8be failed to resubmit (2)
[595468.886390] Bluetooth: hci1: RTL: examining hci_ver=0a hci_rev=dfc6 lmp_ver=0a lmp_subver=d922
[595469.093226] Bluetooth: hci1: RTL: examining hci_ver=0a hci_rev=000b lmp_ver=0a lmp_subver=8761
[595469.095219] Bluetooth: hci1: RTL: rom_version status=0 version=1
[595469.095243] Bluetooth: hci1: RTL: loading rtl_bt/rtl8761bu_fw.bin
[595469.095594] Bluetooth: hci1: RTL: loading rtl_bt/rtl8761bu_config.bin
[595469.095846] Bluetooth: hci1: RTL: cfg_sz 6, total sz 30210
[595469.364213] Bluetooth: hci1: RTL: fw version 0xdfc6d922
[595469.466186] Bluetooth: MGMT ver 1.23

bicaluv, 2025-07-17:
What raspberry pi are you using? Looks like you added BT via usb?
I'm using PI Zero 2 W which has BT onboard and it works well.

jkerekes99, 2025-07-24:
@bicaluv I'm using a Pi 2b with a USB Bluetooth 5/LE adapter.
Was merely trying to use parts I already had.

But it looks like this is causing me issues.
I have just ordered a Pi 5b with built-in Bluetooth, I will try with this; fingers crossed, there will be no more issues with the new Pi.