wimaha/TeslaBleHttpProxy #134: Big issues since few weeks. Canot connect to car

Tesla234987234sdf, 2025-09-06:
Hello,

since few weeks i have big issues. Canot connect to car most of the time. Sometimes works. 

<html><body>

  | "Vehicle is not in range: ble: failed to scan for LRW3E7EK1NC510698: context deadline exceeded"
-- | --

</body>
</html>

Tesla234987234sdf, 2025-09-06:
2025/09/06 19:25:08 INFO TeslaBleHttpProxy 2.1.3 is loading ...
2025/09/06 19:25:08 DEBU LogLevel set to debug
2025/09/06 19:25:08 INFO Env: httpListenAddress=:8080
2025/09/06 19:25:08 INFO Env: cacheMaxAge=5
2025/09/06 19:25:08 INFO Env: scanTimeout=30
2025/09/06 19:25:08 DEBU PrivateKeyFile loaded PrivateKeyFile=key/private.pem
2025/09/06 19:25:08 INFO BleControl initialized
2025/09/06 19:25:08 INFO TeslaBleHttpProxy is running!
2025/09/06 19:25:09 DEBU Waiting for next command ...
2025/09/06 19:25:13 DEBU Received HTTP request Handler=BodyControllerState Method=GET Endpoint=/api/1/vehicles/LRW3E7EXXXXXXXXX/body_controller_state Client=192.168.178.30:51132
2025/09/06 19:25:13 DEBU Scanning for vehicle ...
2025/09/06 19:25:28 DEBU Response Command=body-controller-state Status=503 Result=false Reason="Vehicle is not in range: ble: failed to scan for LRW3E7EXXXXXXXXX: context deadline exceeded"
2025/09/06 19:25:33 can't accept: listner timed out
2025/09/06 19:27:46 DEBU Received HTTP request Handler=BodyControllerState Method=GET Endpoint=/api/1/vehicles/LRW3E7EXXXXXXXXX/body_controller_state Client=192.168.178.30:51288
2025/09/06 19:27:46 DEBU Scanning for vehicle ...
2025/09/06 19:27:53 DEBU Beacon found LocalName=S50145329c4793aa7C Address=ac:4d:16:0e:7c:81 RSSI=-40
2025/09/06 19:27:53 DEBU Creating vehicle object ...
2025/09/06 19:27:53 DEBU Connecting ...
2025/09/06 19:27:53 DEBU Starting VCSEC session ...
2025/09/06 19:27:53 INFO Executing command Command=body-controller-state Body=map[]
2025/09/06 19:27:54 INFO Successfully executed Command=body-controller-state Body=map[]
2025/09/06 19:27:54 DEBU Response Command=body-controller-state Status=200 Result=true Reason="The request was successfully processed."
2025/09/06 19:56:11 DEBU Received HTTP request Handler=BodyControllerState Method=GET Endpoint=/api/1/vehicles/LRW3E7EXXXXXXXXX/body_controller_state Client=192.168.178.30:52641
2025/09/06 19:56:11 DEBU Scanning for vehicle ...
2025/09/06 19:56:21 DEBU Beacon found LocalName=S50145329c4793aa7C Address=ac:4d:16:0e:7c:81 RSSI=-37
2025/09/06 19:56:21 DEBU Creating vehicle object ...
2025/09/06 19:56:21 DEBU Connecting ...
2025/09/06 19:56:21 DEBU Starting VCSEC session ...
2025/09/06 19:56:21 INFO Executing command Command=body-controller-state Body=map[]
2025/09/06 19:56:21 INFO Successfully executed Command=body-controller-state Body=map[]
2025/09/06 19:56:21 DEBU Response Command=body-controller-state Status=200 Result=true Reason="The request was successfully processed."
2025/09/06 20:30:34 DEBU Received HTTP request Handler=BodyControllerState Method=GET Endpoint=/api/1/vehicles/LRW3E7EXXXXXXXXX/body_controller_state Client=192.168.178.30:54490
2025/09/06 20:30:34 DEBU Scanning for vehicle ...
2025/09/06 20:30:38 DEBU Beacon found LocalName=S50145329c4793aa7C Address=ac:4d:16:0e:7c:81 RSSI=-41
2025/09/06 20:30:49 DEBU Response Command=body-controller-state Status=503 Result=false Reason="failed to connect to vehicle (A): ble: failed to enumerate device services: ATT request failed: input channel closed: io: read/write on closed pipe"
2025/09/06 20:31:01 DEBU Received HTTP request Handler=BodyControllerState Method=GET Endpoint=/api/1/vehicles/LRW3E7EXXXXXXXXX/body_controller_state Client=192.168.178.30:54490
2025/09/06 20:31:01 DEBU Scanning for vehicle ...
2025/09/06 20:31:05 DEBU Beacon found LocalName=S50145329c4793aa7C Address=ac:4d:16:0e:7c:81 RSSI=-40
2025/09/06 20:31:05 DEBU Creating vehicle object ...
2025/09/06 20:31:05 DEBU Connecting ...
2025/09/06 20:31:05 DEBU Starting VCSEC session ...
2025/09/06 20:31:06 INFO Executing command Command=body-controller-state Body=map[]
2025/09/06 20:31:06 INFO Successfully executed Command=body-controller-state Body=map[]
2025/09/06 20:31:06 DEBU Response Command=body-controller-state Status=200 Result=true Reason="The request was successfully processed."
2025/09/06 20:57:43 DEBU Received HTTP request Handler=BodyControllerState Method=GET Endpoint=/api/1/vehicles/LRW3E7EXXXXXXXXX/body_controller_state Client=192.168.178.30:55763
2025/09/06 20:57:43 DEBU Scanning for vehicle ...
2025/09/06 20:57:44 DEBU Beacon found LocalName=S50145329c4793aa7C Address=ac:4d:16:0e:7c:81 RSSI=-43
2025/09/06 20:57:58 DEBU Response Command=body-controller-state Status=503 Result=false Reason="failed to connect to vehicle (A): ble: failed to enumerate device services: ATT request failed: input channel closed: io: read/write on closed pipe"
2025/09/06 20:58:01 DEBU Received HTTP request Handler=BodyControllerState Method=GET Endpoint=/api/1/vehicles/LRW3E7EXXXXXXXXX/body_controller_state Client=192.168.178.30:55763
2025/09/06 20:58:01 DEBU Scanning for vehicle ...
2025/09/06 20:58:05 DEBU Beacon found LocalName=S50145329c4793aa7C Address=ac:4d:16:0e:7c:81 RSSI=-47
2025/09/06 20:58:06 DEBU Creating vehicle object ...
2025/09/06 20:58:06 DEBU Connecting ...
2025/09/06 20:58:06 DEBU Starting VCSEC session ...
2025/09/06 20:58:06 INFO Executing command Command=body-controller-state Body=map[]
2025/09/06 20:58:06 INFO Successfully executed Command=body-controller-state Body=map[]
2025/09/06 20:58:06 DEBU Response Command=body-controller-state Status=200 Result=true Reason="The request was successfully processed."
2025/09/06 21:25:42 DEBU Received HTTP request Handler=BodyControllerState Method=GET Endpoint=/api/1/vehicles/LRW3E7EXXXXXXXXX/body_controller_state Client=192.168.178.30:57106
2025/09/06 21:25:42 DEBU Scanning for vehicle ...
[... cut ...]
2025/09/06 21:50:46 DEBU Response Command=body-controller-state Status=200 Result=true Reason="The request was successfully processed."
2025/09/06 22:20:40 DEBU Received HTTP request Handler=BodyControllerState Method=GET Endpoint=/api/1/vehicles/LRW3E7EXXXXXXXXX/body_controller_state Client=192.168.178.30:60000
2025/09/06 22:20:40 DEBU Scanning for vehicle ...
2025/09/06 22:20:55 DEBU Response Command=body-controller-state Status=503 Result=false Reason="Vehicle is not in range: ble: failed to scan for LRW3E7EXXXXXXXXX: context deadline exceeded"

bicaluv, 2025-10-03:
Have you tried using it outside of your docker container?  
I compiled it for my raspi and it runs really nicely without any issues (okay, getting 503s but they do not effect operation)

ghlqz, 2025-11-05:
I also had the "context deadline exceeded" error after months of reliable operation. I could not even reconnect the car. 

Then I switched back to version 2.0.3 and so far all is good again. To do so, change the image value in the docker-compose.yml to

image: wimaha/tesla-ble-http-proxy@sha256:8be3f7675036579de89a51338482b6b7cccfce383d076aa2f056579a460e75d5

The hash for the different versions can be found as "digest" on 
[https://hub.docker.com/r/wimaha/tesla-ble-http-proxy](https://hub.docker.com/r/wimaha/tesla-ble-http-proxy)

wimaha, 2025-11-05:
Did you try to increase the scanTimeout?

ghlqz, 2025-11-08:
Some update, it now seems to be a confirmed problem with the car. I found some related discussion in the Tesla friends forum and others confirmed the problem on the evcc discussion list. With current software, Tesla stops charging if a Bluetooth key approaches the car or is present nearby. I think that tesla-ble acts like such a key. 

For the approaching keys a workaround is to disable Bluetooth in the mobile phone, but this is of course no solution for tesla-ble. Thus all that remains is to wait for an update of the car software.

ghlqz, 2025-11-12:
No, but some users of the Tesla friends forum did it and Tesla confirmed a software bug related to charging and Bluetooth. They say it was forwarded to the development department. 

https://tff-forum.de/t/laden-stoppt-bei-verbindung-mit-handyschluessel-bluetooth-bh-sens/401409/40

ghlqz, 2025-11-24:
Received the update to 2025.44.3 which lets me charge via evcc with TeslaBleHttpProxy again. Did not try PV but only fast charging yet. But before the charging always stopped after seconds. In the Tesla friends forum similar reports can be found.